flipperdevices/flipperzero-firmware #2547: HID iClass 13.56 MHz dongles - not currently readable

C0ntr07, 2023-03-29:
### Describe the enhancement you're suggesting.

Will the Flipper every be able to read HID iClass 13.56 MHz dongles?

Maybe I am just doing something wrong, but I am not (obviously) able to read them with any of the current capabilities.

skotopes, 2023-03-29:
Which specific HID type are we talking about? There is a Pisopass application that provides support for some of HF HID tags.

guardianofroin, 2023-03-30:
Well, while Picopass seems to be able to read the cards, so far I don't see an option to emulate it. Just to write it onto another card.

bettse, 2023-04-09:
@guardianofroin I think we're waiting to hear back from @C0ntr07 about reading a specific type.  Emulation is outside the scope of this issue.

bettse, 2023-07-22:
can we close this?  The flipper zero absolutely can read HID iClass, the only limitations are elite/custom keys or SE.  Also, the picopass app has been moved out of this repo, so if the person still has an issue, they should open an issue against https://github.com/flipperdevices/flipperzero-good-faps

skotopes, 2023-07-23:
Yep

closed 2023-07-23 by skotopes