flipperdevices/flipperzero-firmware #2550: Marked keys as not found result in not working card clone

ramondunker, 2023-03-30:
### Describe the bug.

Due to the change in pull request #2476 it's no longer possible to clone the Mifare Classic 1K badges of the brand Hotek (https://www.hotek.nl/en/hotel-door-locks/). After succesfully performing the mfkey32 attack and reading the badge only 30/32 keys are found, whereas without this commit 31/32 keys are found. The NFC dump with 31 "found" keys opens the lock, and the dump with 30 "found" keys does not work with the lock.

### Reproduction

A little hard to reproduce as I tested this in a hotel. I could share the corrected dump privately.

### Target

https://github.com/flipperdevices/flipperzero-firmware/pull/2476

ramondunker, 2023-03-30:
Hi @Astrrra ,

Can I send it to your mailaddress? Don't think I should share a full dump here ;)

Astrrra, 2023-03-30:
@ramondunker, yes, that would be a good idea! Please send it to astra@flipperdevices.com

Astrrra, 2023-04-03:
Hm, I can't reproduce your bug on the latest release firmware. My user dictionary contains the only non-standard key for the card, and I get 31/32 keys read (as expected). Do you have any extra details on reproducing this?

ramondunker, 2023-04-07:
You didn't have to add additional keys? I had to perform mfkey32 to calculate the keys in order to read the sectors.

Astrrra, 2023-04-14:
> You didn't have to add additional keys? I had to perform mfkey32 to calculate the keys in order to read the sectors.

As I said in my previous message, I added the custom key from your dump to my user dictionary, which is equivalent to getting it via mfkey32, just quicker

ramondunker, 2023-04-14:
Ah right, I understand. Maybe I did something wrong. Once I'm in that particular hotel I'll try again!

closed 2023-04-14 by ramondunker