flipperdevices/flipperzero-firmware #2552: `compilesort.hpp` missing from 0.80.1 SDK archive

dcoles, 2023-03-31:
### Describe the bug.

When attempting to generate Rust bindings for the 0.80.1 release using [`flipper-z-f7-sdk-0.80.1.zip`](https://update.flipperzero.one/builds/firmware/0.80.1/flipper-z-f7-sdk-0.80.1.zip), I get an error because `lib/flipper_application/api_hashtable/compilesort.hpp` is not found in the SDK archive.

The [current way bindings are generated](https://github.com/flipperzero-rs/flipperzero/blob/f7de5dfb209a2d93f30e5658132e49d1c2f563df/tools/src/bin/generate-bindings.rs#L106) is by pulling in all public (`+`) headers listed in [`api_symbols.csv`](https://sourcegraph.com/github.com/flipperdevices/flipperzero-firmware@0.80.1/-/blob/firmware/targets/f7/api_symbols.csv?L114). Thus if a file is missing from `flipper-z-f7-sdk-x.x.x.zip`, it won't be possible to generate bindings.

This can be worked around by manually copying the file from the git tree.

### Reproduction

1. Download [`flipper-z-f7-sdk-0.80.1.zip`](https://update.flipperzero.one/builds/firmware/0.80.1/flipper-z-f7-sdk-0.80.1.zip) from update.flipperzero.one
2. Run `generate-bindings` script [as specified in the `flipperzero-rs` documentation](https://github.com/flipperzero-rs/flipperzero/blob/f7de5dfb209a2d93f30e5658132e49d1c2f563df/docs/updating-sdk.md#current-process)
3. Observe the script fails due to being unable to find `lib/flipper_application/api_hashtable/compilesort.hpp`

### Target

f7

### Logs

```Text
PS> target\debug\generate-bindings.exe ..\..\flipperzero-firmware\build\f7-firmware-D\sdk
Generating bindings for SDK 00140000
warning: optimization flag '-fsingle-precision-constant' is not supported [-Wignored-optimization-argument]
C:\Users\dcoles\workspace\flipperzero-rs\tools\header.h:114:10: fatal error: 'lib/flipper_application/api_hashtable/compilesort.hpp' file not found     
clang diag: warning: optimization flag '-fsingle-precision-constant' is not supported [-Wignored-optimization-argument]

clang diagnosed error: C:\Users\dcoles\workspace\flipperzero-rs\tools\header.h:114:10: fatal error: 'lib/flipper_application/api_hashtable/compilesort.hpp' file not found
thread 'main' panicked at 'failed to generate bindings', src\bin\generate-bindings.rs:217:13note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace 
```
```

### Anything else?

I could skip including any headers listed in `api_symbols.csv` if the matching header isn't found in the SDK directory, but this might result in bindings not being generated for all public symbols.

Let me know if that should be the correct behaviour.

hedger, 2023-03-31:
Will be fixed in next release.

hedger, 2023-04-07:
@dcoles should be fixed in dev & latest RC. SDK structure has changed too.

dcoles, 2023-04-07:
@hedger Thanks! I'll also take a look at whether the new SDK structure will be a problem (at worst, I might have to tweak the binding generation script).

hedger, 2023-05-22:
Should be fixed a few releases ago.

closed 2023-05-22 by hedger