flipperdevices/flipperzero-firmware #2556: Read and emulate magstrip cards

rhysperry111, 2023-03-31:
### Description of the feature you're suggesting.

I'm particularly interested in using the Flipper Zero to read and emulate magnetic stripe cards as I manage a building that uses them, and it would be useful to be able to test and store many cards.

I have seen [this post](https://samy.pl/magspoof/) suggest it's possible with the 125KHz antenna out of the box, and the general idea of how it works can be seen in [MagSpoof](https://samy.pl/magspoof/). It also seems that there's a [similar feature request](https://github.com/DarkFlippers/unleashed-firmware/issues/193) for DarkFlipper's unleashed-firmware. If this does just work out of the box, it would be useful to know the best way of doing it and possibly mention it in the wiki. If it would be more practical to have this implemented separately as a new feature, that would be nice to see too.

Sadly I don't have much development experience, and I'm still waiting for my Flipper Zero in the mail, so I can't test it myself, but I'd be happy to help in any way I could.

skotopes, 2023-04-04:
There is an implementation that was made 2 years ago for hackathon. Shouldn't take much time to port it for new firmware.

closed 2023-04-04 by skotopes

rhysperry111, 2023-04-04:
The one from the hackathon used an external antenna. Should this issue not stay open until something is implemented?

skotopes, 2023-04-04:
They told me that they've tested internal one too, also was working.

skotopes, 2023-04-05:
Regarding closing issue: this sounds like idea for community project, we are not planning to implement it.