flipperdevices/flipperzero-firmware #2558: WeatherStation not decoding TX141TH-BV2

ntroup, 2023-04-01:
### Describe the bug.

I have a brand-new FlipperZero, and am testing the WeatherStation with a TX141TH-BV2 Temperature Sensor.  This is a supported sensor, but it is not being decoded at all, despite the FlipperZero receiving the signal.

At the same time, I am running rtl_433, and it can properly decode the signal into the bit stream **0xEFC2D7229C**.  The SensorID is the first byte, 0xEF, and that matches the below decode from rtl_433

[image]

I recorded the same signal with the 'READ RAW', and put it into the Pulse Plotter (with settings PWM, Short 209, Long 419, Sync 810), and the Pulse Plotter decodes the signal to **0x103D28DD63**.  The short/long values are close to what I see in rtl_433 and the lacrosse_tx141thbv2.c.  

This is very much a bit flip of what the rtl_433 (and correct) signal is.  

In the rtl_433 code, I see a call to 'bitbuffer_invert' to invert the stream, but I don't see that in the WeatherStation code.

Additionally, and even stranger, I see both FlipperZero forum posts, and YouTube examples, using the same sensor working correctly, out of the box.

So, is there something with my FlipperZero?  Something with my settings?  Or something with the code?  Should the short/long values be flipped to invert the bits?  

[TX141THBV2 Failed Decode.sub.txt](https://github.com/flipperdevices/flipperzero-firmware/files/11129873/TX141THBV2.Failed.Decode.sub.txt)

### Reproduction

1) Open WeatherStation and start receiving
2) Press Transmit button on the TX141TH-BV2
3) RSSI meter shows transmission
4) No signal recorded

Attached .sub file was recorded via 'READ RAW'

ntroup, 2023-04-01:
Photo From the Sensor:
[image]

Skorpionm, 2023-04-02:
[image]
you have a different protocol. and the bitness does not match the duration with what is added. possibly some kind. I will watch

Skorpionm, 2023-04-02:
https://github.com/flipperdevices/flipperzero-firmware/pull/2559
check plz

ntroup, 2023-04-02:
Looks good.  Great work!
[image]

closed 2023-04-03 by ntroup