flipperdevices/flipperzero-firmware #2562: Skylanders real-time write issue

Mikeonut, 2023-04-04:
### Describe the enhancement you're suggesting.

While Skylanders can work after being converted from .bin to .nfc files, they can only currently function on the Switch in a proper manner due to a writing oversight. In games that require the portal to be in use, whenever the portal writes to the figure, it will fail and re-set the figure to its original state. No progress can be made on the figure files. The portal writes to the figure when applying xp, gold, hats, upgrades, nicknames, basically everything you can do with an actual figure.

There needs to be a way to allow for these changes to be applied in real time so that issues like this in game do not occur, otherwise the only games that this device will work on are the 3ds games and the Nintendo switch version of Imaginators. This issue also probably affects Disney Infinity and Lego Dimensions figures as well.

Mikeonut, 2023-08-02:
Checking back this is still an issue. Most recent update 0.87.0, changed how NFC works and now characters cant be emulated at all. In game it just shows their element and then fails to read-resulting in a loop. Assume problem above still exists as well.

z3razerviper, 2023-08-07:
I am using it for amiibo and it keeps hard locking its only gotten worse in the new firmware.

doomwastaken, 2023-11-03:
@Astrrra could you look at this.

skotopes, 2024-01-12:
@Mikeonut @z3razerviper please check latest release

Mikeonut, 2024-01-13:
Hi, latest release fixed the most recent bugs where there were emulation problems but real time read and write is still problematic.

skotopes, 2024-01-14:
@doomwastaken can you please reproduce it? And then transfer this issue to @gornekich

skotopes, 2024-07-07:
@gornekich @doomwastaken ?

Mikeonut, 2024-07-08:
Issue is not solved, and the newest update seems to have regressed the changes from the last few updates, making the figure dumps impossible to read directly from the flipper.

skotopes, 2024-07-09:
> Issue is not solved, and the newest update seems to have regressed the changes from the last few updates, making the figure dumps impossible to read directly from the flipper.

you need to provide more details: 
- fw version
- what exactly you doing
- what exactly you see
- logs from the device
- communication logs, if you have proxmark

Mikeonut, 2024-07-09:
- Firmware 0.103.1
- Emulating Skylander NFC Dump (Mifare 1k)
- Correct element for Dumped figure appears in game, but then suddenly vanishes, like constantly placing and removing legit figure multiple times (What used to happen was dumped figure would load in game, but whenever coins were collected or figure was modified in any way, it would automatically de-load and then re-load without the changes applied.
- No Logs produced, device doesnt error out in any way
- Do no have proxmark

skotopes, 2024-07-09:
Go to Settings->System->Log Level set it to debug.
Read logs through cli or wifi board, more info: https://docs.flipper.net/development/hardware/wifi-developer-board/reading-logs