flipperdevices/flipperzero-firmware #2563: U2F does not work with Facebook

JulyIghor, 2023-04-05:
### Describe the bug.

YubiKey successfully works with Facebook, but Flipper Zero can't be added in any case.

### Reproduction

1. Open Safari, macOS
2. Visit facebook.com
3. Try to add Flipper Zero's U2F as a security key
4. You get infinite load instead of the key to be added

### Target

U2F works with Facebook

JulyIghor, 2023-04-05:
Also it does not work with GitHub
[image]

JulyIghor, 2023-04-05:
[image]

JulyIghor, 2023-04-05:
[image]

ChlorideCull, 2023-04-13:
As far as I can tell the issue is that the U2F application only implements CTAP1, while most sites require CTAP2 for new registrations at this point.