flipperdevices/flipperzero-firmware #2577: NFC: Emulating of SAVED Mifare Classic not working

AlexMilender, 2023-04-12:
### Describe the bug.

Emulation of Mifare Classic produces different results depending on if the Mifare Classic fob has been saved or not.

I am able to successfully read a Mifare Classic key fob using NFC read functionality.
If I do NOT save the read, but instead emulate it directly after the read. The emulation works as expected and i am able to open my key fob lock

If I DO save the read and then try to emulate that saved read. The emulation does not work as expected and I am unable to open my key fob lock

### Reproduction

Successful able to open keyfob lock
1. NFC
2. Read
3. More
4. Emulate

Unsuccessful in opening keyfob lock
1. NFC
2. Read
3. More
4. Save the read
5. NFC
6. Saved
7. Naviatege to saved read from step 4
8. Emulate

### Target

Flipper Zero Release 0.80.1

bettse, 2023-04-12:
What brand is the lock?  I'd like to see if this is a problem for _all_ emulation, or against a specific lock.

doomwastaken, 2023-04-12:
Going to research it now on 0.81.1-rc

doomwastaken, 2023-04-12:
If you could provide additional details about lock and what card you are using, that would be helpful, tried Mifare Classic 1k and 4k and was not succesful so far

Astrrra, 2023-04-12:
Also, having the full card dump will help out a lot

Indigo94, 2023-04-12:
I’m experiencing the same issue as OP. The reader is a Schlage lock

Panduhsaur, 2023-04-12:
I have the same issue as OP.

Here's the Schlage model: https://www.schlage.com/en/home/products/BE467GRWFFF.html?fbclid=IwAR29B_GfcXJUn_cKD3ezzMMTAhIU_PCkSxDqYFbRcEwzHy6_i11rw30EsHI

gornekich, 2023-04-12:
Could you please try to reproduce the issue on previous release?

AlexMilender, 2023-04-12:
> What brand is the lock? I'd like to see if this is a problem for _all_ emulation, or against a specific lock.

The lock brand is Schlage. The same one @Panduhsaur mentioned (https://www.schlage.com/en/home/products/BE467GRWFFF.html)

Key fob is a Schlage 9651t

Indigo94, 2023-04-13:
> Could you please try to reproduce the issue on previous release

How can I do this? I attempted to download the .zip and the tar.gz file, but when I go on qFlipper (windows) -> install from file -> and select one or the other, the downgrade fails

AlexMilender, 2023-04-13:
> Could you please try to reproduce the issue on previous release

@gornekich That did it! I used fbt to build and flash version 0.79.1 and i am now able to save and then emulate the key fob

AlexMilender, 2023-04-13:
Update: After successfully saving and emulating the keyfob in version 0.79.1. I then updated the firmware to version 0.81.1 and could emulate the saved keyfob with success.

Indigo94, 2023-04-13:
As the previous user mentioned, downgrading to version 0.79.1 fixed the issue.
For users that would like to do this, do the following:

1. Clone the following repository using the following command:
`git clone --recursive --branch 0.79.1 https://github.com/flipperdevices/flipperzero-firmware.git`
2. Go into the new 'flipper-zero firmware' folder
3. run the fbt command:
`./fbt`
4. Connect your Flipper and run the following command:
`./fbt flash_usb`

doomwastaken, 2023-04-15:
Could you please us dump of a key with latest release and 0.79.1, we would like to compare files. You could mail them to k.volkov@flipperdevices.com. 

If by chance any of you have proxmark, then let us know a way to contact you

m-kozlowski, 2023-04-17:
Here are some of my observations:
I have two mfc cards, one for workplace (A) and one for public transport (B)

(A) is purely UID based, all sectors are filled with zeroes with FFFFFFFFFFFF as both keys and 000/000/000/001 access bits
(B) has non-default keys and is using sectors 0-8. Moreover single block of sector 4 has read and write disabled by access bits (111)

Results of emulating card A with flipper:
- did not work with door sensors at my workplace 
- it did work with storage locker (some kind of battery powered reader)
- Mifare Classic tool throws "NFC read error"
- ACR122U 
- - nfc-list lists card properly
- - trying to read card with mfoc or nfc-mfclassic immediately says "no tag was found"

Results of emulating card B:
- Mifare Classic tool detects card, show it's UID but i'm unable to read card (guess that's to be expected with the way emulation works)
- ACR122U 
- - nfc-list works
- - trying to read card with mfoc or nfc-mfclassic also didn't work, but this time nfc-mfclassic displayed card details before throwing "tag disappeared" error.

There are no notable differences between 0.79.1 and 0.81 dumps in my case.
For both cards it's limited to header:
```
@@ -1,9 +1,10 @@
 Filetype: Flipper NFC device
 Version: 3
-# Nfc device type can be UID, Mifare Ultralight, Mifare Classic
+# Nfc device type can be UID, Mifare Ultralight, Mifare Classic, Bank card or ISO15693
 Device type: Mifare Classic
 # UID, ATQA and SAK are common for all formats
 UID: .. .. .. ..
+# ISO14443 specific fields
 ATQA: 00 04
 SAK: 08
 # Mifare Classic specific data

```

AlexMilender, 2023-04-22:
> Could you please us dump of a key with latest release and 0.79.1, we would like to compare files. You could mail them to [k.volkov@flipperdevices.com](mailto:k.volkov@flipperdevices.com).
> 
> If by chance any of you have proxmark, then let us know a way to contact you

A key dump of the 2 versions you mentioned have been emailed to the link you provided. Unfortunately I do not own a proxmark

davenukem, 2023-05-20:
I can confirm this issue is still ongoing on most recent firmware. When reading an NFC fob and emulating from the initial read it will function. 

As soon as you save that read, the emulation will no longer work. I have the same Schlage device as the one previously listed.

meppss, 2023-06-04:
> Currently this fix is blocked by NFC refactoring process. I will update this issue once refactor is done, cannot give any ETA.

I'm confirming that this is still broken in the latest release [0.84.1]. Attempted on a freshly updated flipper.

warnerlowe, 2023-06-05:
I was just wondering about this issue. Glad it's being worked on.

skotopes, 2023-06-09:
fixed in dev

meppss, 2023-06-09:
> fixed in dev

Can you share the PR this was fixed in? Thanks!

doomwastaken, 2023-06-09:
> > fixed in dev
> 
> Can you share the PR this was fixed in? Thanks!

https://github.com/flipperdevices/flipperzero-firmware/pull/2620

davenukem, 2023-06-09:
> fixed in dev

Just installed the .85 RC. Completely possible I'm doing something wrong but this seems to actually eliminate the ability of the Flipper to even emulate the MIFARE classic initially. In the current firmware, you can emulate your MIFARE classic NFC after an initial read but upon saving it will no longer function if you attempt to emulate. 

On the RC firmware .85, even the initial emulation fails for me now. 

Rolling back to the .84.2 firmware initial emulation once again functions, saved emulation fails. 

Not sure if this ties into the NFC refactoring you mentioned - but the issue doesn't appear to be fixed in .85

davenukem, 2023-06-09:
I read 16/16 sectors and get 30/32 keys (same as firmware .84.2). I ran through the entire card read process again and detected reader to get nonces again as well (on both firmwares). Ran through the same process on both firmwares. 

Firmware .85 will not even succeed at initial emulation, .84.2 will. 
Neither will allow an emulated saved read to function.

davenukem, 2023-06-09:
> > I read 16/16 sectors and get 30/32 keys (same as firmware .84.2). I ran through the entire card read process again and detected reader to get nonces again as well (on both firmwares). Ran through the same process on both firmwares.
> 
> And you ran the Mfkey attack and scanned the key again after?

Yes. This is something I would love to work… so I went through the whole process twice to make sure it wasn’t me messing up.

Astrrra, 2023-06-13:
@davenukem can you please send all the files needed to reproduce your problem? Please include the card dumps from 0.84.2 and from 0.85, the nonce files from running Detect reader, along with the sector numbers and key letters to which the reader is trying authenticate (e.g. Sector 8 Key A).

If your card contains private data - you can send it to me directly: astra@flipperzero.one. Also, do you have a proxmark3? Having a trace file from it would greatly simplify things.

davenukem, 2023-06-13:
I’ll try and get this over in the next day or so.

I don’t have a proxmark (but maybe this is a reason to finally snag one!).

Expect an update shortly.

warnerlowe, 2023-06-13:
I see your cellphone number.

meppss, 2023-06-19:
I tried this same process on the latest version. Emulation no longer functions after retrieving the keys, but before saving.

I've also tried saving the read and then doing a detect reader. That also does not work (the reader doesn't detect the NFC emulation).

However after completing a read, when performing a "detect reader" in that same menu, while collecting the nonces the reader detects the emulated signal and unlocks the door. This only occurs during the "detect reader" function while nonces are being collected.

I'm performing this on the latest stable 0.85.2

davenukem, 2023-06-24:
> I tried this same process on the latest version. Emulation no longer functions after retrieving the keys, but before saving.
>
> I've also tried saving the read and then doing a detect reader. That [...]

I just sent an email with the key that is generated and the user dictionary I create from the nonces on the reader to Astrrra. Unfortunately wasn't able to roll back to old firmware due to some time constraints on my end. Upon doing it again today I can confirm what Meppss was saying. The detect reader function emulation actually does trigger the device while acquiring nonces and not when you try and emulate from the same menu post reading. Overall, extremely odd behavior and seems to be some kind of disconnect between how NFC reads are being saved and passed to the emulate function immediately post read.  

Hopefully this helps move this issue along.

meppss, 2023-07-20:
Confirming this is fixed as of 0.86.2

zhiyan114, 2023-07-21:
Seems like I'm still having issue with this as of the latest release (0.87.0). Though could be a fluke, so I'll try again next week.

tomaszkoperski, 2023-08-01:
I'm having the same issue. I successfully got the keys to a Mifare Classic 1K card (Keys found: 32/32, Sectors read: 16/16), but I'm unable to open the lock (either through directly emulating the card after read or after the card is saved). I'm using FW  0.87.1.

bwachter, 2023-08-20:
Ran into this issue yesterday as well. Was originally assuming a corrupted card dump, but converting the flipper nfc file to proxmark bin and uploading it to a proxmark with HF_MFCSIM let me correctly emulate the card there.

skotopes, 2023-08-22:
we need exchange dump (can be sniffed with proxmark)

doomwastaken, 2023-11-03:
@bwachter @AlexMilender could you check this again on latest firmware?

hmvs, 2023-11-04:
Now it is working! Thanks

AlexMilender, 2023-11-04:
> @bwachter @AlexMilender could you check this again on latest firmware?

Yes, it is working in v0.94.1. Thanks!

closed 2023-11-08 by doomwastaken