flipperdevices/flipperzero-firmware #2578: New RFID Card Protocol Support

zigad, 2023-04-12:
### Describe the enhancement you're suggesting.

I have a access card that is completely white and has no markers just few dots.

[image]

It can be read with this reader: [GE Security 430222003 Model T-100 reader, Black, mini-mullion, GE Proximity (ProxLite, Entree Prox), Wiegand only 13](https://www.surveillance-video.com/security-430222003.html) So I guess the protocol for this card is ProxLite or Entree Prox.

FCC ID of the reader: [R2L1050MINI](https://fccid.io/R2L1050MINI)

This is the RAW reading of the card: [RfidRecordRaw.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11208071/RfidRecordRaw.zip)

Akury83, 2023-04-16:
From those dots, this may be an HID card, have you tried reading with the picopass app?

zigad, 2023-04-16:
> From those dots, this may be an HID card, have you tried reading with the picopass app?

Tried it also with picopass app but no luck

Astrrra, 2023-06-02:
I've looked at the RAW recordings, the ASK one looks like it has a proper signal on it! We'll try to implement this protocol if possible.