flipperdevices/flipperzero-firmware #2588: NFC 13,56MHz can't be readed

javiercalzadaquesada, 2023-04-15:
### Description of the feature you're suggesting.

Hello team i have this tag but can no read with flipper https://www.golmar-seguridad.es/productos/evo-tag. Any one have any idea how to read with flipper?

Astrrra, 2023-06-01:
Hi! Does the Flipper not react to the tag at all, or does it display at least the serial number (UID) of the tag?

mgrybyk, 2023-06-13:
@Astrrra I have issues with golmar devices as well. I'm using https://www.golmar-seguridad.es/productos/tagdoor-mf-plus#product that seems to use 13.56MHz as well and supports anti-copy encryption: AES-128 (according to the website).

- I can read my key, see [key.nfc.txt](https://github.com/flipperdevices/flipperzero-firmware/files/11729188/key.nfc.txt)
- I've tried to use the detect reader feature and [Mfkey32v2](https://lab.flipper.net/nfc-tools). It works really slow with the reader and I have doubts if it works correctly.

```
0b8c9c1fa4a5
3d483aa5a4a5
a0a1a2a3a4a5
```

However, the emulation doesn't work.
I'm more than happy to help investigating the issue with your guidance.

Thank you!

Astrrra, 2023-06-13:
@mgrybyk Hi! This is a completely different tag, even though it's branded by the same seller. The chip inside is Mifare Plus, and in your case it looks like it's in SL1, which makes it act like Mifare Classic chip, which is supported.

This issue is for EVO tags specifically, please open a separate issue for your problem, as apart from being sold by the same seller, those two are not related in any way.

Also, please include what specific sectors the keys recovered by Detect Reader are from (e.g. Sector 3 Key A: A0A1A2A3A4A5, Sector 7 Key B: 0B8C9C1FA4A5, etc), along with everything you've included in this issue.

Astrrra, 2023-07-03:
Closing this issue as stale. If you still need help with this type of tag - please open a new one.

closed 2023-07-03 by Astrrra