flipperdevices/flipperzero-firmware #2589: Picopass app not reading elite keyed cards anymore.

Akury83, 2023-04-16:
### Describe the bug.

Hi all,

I just upgraded from ver 0.78.1 to ver 0.81.1 and I’ve found that the picopass app no longer reads elite keyed cards, even ones keyed with the keys in dictionary files. The old version of the app used to read elite keyed cards with the default keys..

Any help would be greatly appreciated with this one, cheers!

### Reproduction

1. Open picopass app
2. Read standard keyed card using ‘Read Card’ function - works
3. Read standard keyed card using ‘Elite Dict. Attack’ function - works
4. Read elite keyed card (default AEA6… key) using ‘Read Card’ function - fails
5. Read elite keyed card (default AEA6… key) using ‘Elite Dict. Attack’ function - fails

### Target

PicoPass app

### Anything else?

I have tried this on software versions 0.80.1, 0.81.1 and the latest dev build 37fb330b, none seem to work

hedger, 2023-04-16:
@bettse check this out please

bettse, 2023-04-16:
Will do!

bettse, 2023-04-16:
@Akury83 since using the standard key (AEA6..) with the elite KDF is unusual, can you confirm that you copied the standard key to your user elite dictionary file (`SD Card/apps_data/picopass/assets/iclass_elite_dict_user.txt`)?

bettse, 2023-04-16:
I used a proxmark to craft a card that used the standard key, but was diversified using the elite KDF.  
I copied the standard dict, renamed as `iclass_elite_dict_user.txt` and uploaded it, which allowed "Read Card" to work successfully.  I did see a Read Error for the Elite Dict Attack, which I'm going to investigate.

bettse, 2023-04-16:
I pushed a PR that should fix it, based on my testing.

closed 2023-04-17 by skotopes