flipperdevices/flipperzero-firmware #2597: Flipper cannot update sector trailers on Mifare Classic 1k

kulverstukas1, 2023-04-19:
### Describe the bug.

I noticed this what I think is a bug.
I tried with a blank regular Mifare Classic 1k 4 byte UID card with factory keys and factory access bits.

I wrote everything down in the forums but I'm also posting here for more exposure (prolly should of posted here in the first place though)
https://forum.flipperzero.one/t/flipper-cannot-update-sector-trailers-on-mifare-classic-1k/15128

Maybe someone can confirm that this is really a bug?

### Reproduction

1. Read and Save the card dump
2. Edit dump with Flipper Android app
3. Save and synchronize (also verify on the Flipper itself that the file is modified)
4. Write to Initial card
5. Read and see that the keys haven't changed

### Target

NFC / Mifare Classic 1k

Crsarmv7l, 2023-04-20:
AFAIK, you can only write to a Magic Mifare gen1a, not a regular "blank"

kulverstukas1, 2023-04-21:
"Magic" cards differ from regular in a way that you can change the UID.

Astrrra, 2023-06-01:
Yes, the "Write to initial card" feature assumes that the keys remain the same (otherwise the complexity grows very significantly), so it won't be able to reflect the key changes.

You can use the NFC Magic app to write arbitrary data to cards, though only the Magic cards are supported for that.

closed 2023-06-07 by Astrrra