flipperdevices/flipperzero-firmware #2599: Add support for gen2a MFC

optimus10ro, 2023-04-20:
### Description of the feature you're suggesting.

Add the option to Write, wipe, and check for a mfc gen2a in the NFC magic app

quantum-x, 2023-04-29:
Gen2a / FUID One Time / "One Time Write" UID cards can only have their UID changed one time.

Other than that, they identical to Gen2 (direct-write) tags. 
They do not support "unlocked writing / wiping" (like Gen1a tags).

There's no easy way to determine if a card is Gen2a : they often have a generic factory-issued UID, and sometimes reveal themselves with RATS. 

Once written, to 'wipe' the card, you need to know the keys to reset the data (except block0) to " blank " values - if the ACL allows it.

Check out:
https://lab401.com/blogs/academy/know-your-magic-cards (Quick overview of magic cards)
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/magic_cards_notes.md (In depth overview)

Kanaduchi, 2023-05-29:
Is this request planning to take into work?

Currently there are a lot of gen2 tags with one-time writable UID to block 0 that cannot be written by Flipper. Example - https://pt.aliexpress.com/item/1005005052138469.html

To copy data from Mifare Classic 1K to this tag I need firstly update UID with proper UID by MCT android application and only after that I can write dump by Flipper to this card

Kanaduchi, 2023-05-29:
> What do you mean "one-time writable"? That's regular gen2.
thanks for the remark. I mean gen2 support

team-orangeBlue, 2023-05-29:
Other than that, here's a visualization of what the flipper can do:

Gen1A magic - supported
Gen1B magic - rare, no support but possible
Gen2 magic - pending support*
Gen3 magic - supported via CLI (hint: `nfc apdu 90FBCCCC07[UID`)
Gen4 magic - supported

Others..
UFUID / Magic85 chip: need to add 0xE000/0xE100 command
GDM: need to add UFUID stuff and custom auth commands
Supercards: depends

The asterisk is my idea as to how gen2 could be added.
I would say that the flipper lacks a decent storage management interface, or a way to edit it.
So we could allow editing a dump on the flipper and using that, implement gen2 support. As well as better MFC support.

sfjuocekr, 2023-06-27:
DirectWrite cards should be writeable, but currently are not.

theMacLady, 2023-07-28:
Just wanted to bump this feature request and ask if there are any updates?

sfjuocekr, 2023-07-30:
I was looking into this and tried to strip the Gen1 magic app to make it "Gen2".

The Gen2 cards can be written, you can "just" sent a write command to the block and it "should" accept it. From my testing however you need the sector keys to make it work, this means you need to scan the card first to obtain its keys and these keys can then be used to write the Gen2 card with new data.

I haven't worked on this the past weeks much, but might take a look at it again in the future.

theMacLady, 2023-07-30:
To clarify - the gen2 write capability is there, just janky. So you have to

sfjuocekr, 2023-07-30:
You can not do this currently, the hardware can.

You would have to make changes to the NFC Magic app to sent a write command to block 00, for the sector data you NEED the current keys of the Gen2 card so you can write new values to them.

mmiller7, 2023-10-06:
I don't know about "one time writable" UID, I've got a UID-changable ring that as near as I can find is a Gen2 type magic Mifaire Classic 1K chip from China...and I have absolutely written more than once changing it with a Proxmark3

But yes, it would be really nice if the Flipper could write UIDs to UID-changable other than the Gen 1/1a/4 cards.  I'm interested in this feature.

team-orangeBlue, 2023-10-07:
There are some "one time write" tags, the one you will know is FUID aka
China's variant. TLDR: Gen2, default UID is AA56C396, manufacturer data is
null, all anticollision from block 0, can be detected if you want to.

Others are... Russian OTP 2.0, but it's being discontinued. It's an OTP
Gen1a. That's it, really.

Kanaduchi, 2023-11-05:
@Astrrra Hello! Do you have any updates? As I know NFC refactoring was done. Is it time to return to this task?

senpng, 2024-03-01:
https://github.com/flipperdevices/flipperzero-good-faps/pull/143

Can use this pr, and modify it to adapter latest firmware.

I tested this and it can write data to the gen2.

skotopes, 2024-04-14:
Fixed in latest version.

closed 2024-04-14 by skotopes