flipperdevices/flipperzero-firmware #2600: another unknown RFID

VxTedxV, 2023-04-20:
### Description of the feature you're suggesting.

Hi, as described in the new documentation, here is :
- ask file
- psk file
- photo of the card
... all in the [Archive.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11289258/Archive.zip) file

Thanks a lot for all your efforts !

skotopes, 2023-06-06:
ASK file is empty, PSK - no meaningful data. Do you have proxmark?

VxTedxV, 2023-06-06:
no :'(

VxTedxV, 2023-06-07:
Thanks again to investigate :)
Here is another raw recording of the card.
[Archive.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11680062/Archive.zip)

Astrrra, 2023-06-09:
Hi! I've looked at the new files, and they're still only containing random noise, no signal in sight. It seems that this card is either not 125kHz, or is an active card, which Flipper can't read and emulate. In any case, there's nothing we can do with those files, unfortunately.

VxTedxV, 2023-06-09:
Thanks ! Still interesting.

closed 2023-06-09 by VxTedxV

VxTedxV, 2023-07-25:
@Astrrra 
I think it's not an active card : slim, very short range, at least 3 years lifetime for now.
I tried the "NFC/RFID detector" app (video below). It says LF RFID and in less than 2 sec, shows 94.21 kHz and then multiple frequencies near 99 kHz, and finally 102.42 kHz.
Any clue to understand better ?

https://github.com/flipperdevices/flipperzero-firmware/assets/30288828/55e81cfc-4d64-4323-b9be-74407a42aec6

skotopes, 2023-07-26:
https://www.idmanagement.com/id-cards/rfid-cards/cotag
https://www.proxmark.io/www.proxmark.org/forum/viewtopic.php%3Fid=1258.html

We do not support them currently, but proxmark got some kind of support. We may port it in future.

skotopes, 2023-07-26:
Or maybe someone from community will do it first ;-)