flipperdevices/flipperzero-firmware #2609: Fix saving selected settings in Sub-GHz

krolchonok, 2023-04-24:
### Describe the bug.

When you select the settings in "Config" in Read - Sub-GHz and then exit or view the signal information, the "Modulation" setting switches to the standard one.

### Reproduction

EN (ya.translate):
1. Log in to SubGhz - read
2. Catch the signal (Any)
3. After the capture, go to Config
4. Change the modulation to another
5. Go back
6. Open the signal information
7. Go back to the scan menu
8. Check which modulation is worth it, whether it has been reset or not.

Rus:
1. Зайти в SubGhz - read
2. Поймать сигнал (Любой)
3. После поимки зайти в Config
4. Поменять модуляцию на другую
5. Вернуться назад
6. Открыть информацию о сигнале
7. Вернутся обратно в меню сканирования
8. Проверить какая модуляция стоит, сбросилась она или нет.

krolchonok, 2023-04-25:
Thank you. The scenario I see is to find a pair of signals on one modulation, switch to another modulation, continue searching, view information about the old signal without stopping the search on the new modulation.

paulvantuyl, 2023-05-26:
I'm having a similar problem where I adjust the config, go back, scan, and then can't save results. Also, changing the config does not save – it defaults back to 433.92 each time.