flipperdevices/flipperzero-firmware #2612: PicoPass: Elite Dict.Attack with no card present

doomwastaken, 2023-04-25:
### Describe the bug.

Starting the attack promts you with "Lost the tag!" menu, additionally if you press "Skip", you get "Read Failed" with text not fitting in the screen. 

### Reproduction

No card next to flipper. 

- Open PicoPass
- Elite Dict.Attack
- "Lost the tag!"
- [Optional] Skip
- Text overflow

Expected behavior: 
1. Promt for card placement
2. Text either scrolling / compact view / overflow to second line

[image]
[image]

### Target

f7

doomwastaken, 2023-04-25:
@bettse could you please check it out

bettse, 2023-04-25:
ugh.  yeah, not the first time the variable width font has bit me.  I won't be able to open a PR until this evening at the earliest

doomwastaken, 2023-04-25:
Oh, not in a rush, we won't be able to fit into this release anyway, so 2 weeks to work with. Thank you !

bettse, 2023-04-25:
@doomwastaken Thank you for reporting this.  You're helping us make it better!

closed 2023-05-03 by skotopes