flipperdevices/flipperzero-firmware #2618: Flipper Zero cannot read RFID at all. Attached RAW file.

LuciferShadows, 2023-04-26:
### Description of the feature you're suggesting.

Hi,

I've added the raw file of the card. I'd like to know the type and any details.
My flipper zero doesn't seem to be reading RFID at all. NFC seems to be okay.
I'd like to know the format of the attached RAW file to see if this is something the flipper zero should already be able to read and emulate.

### Anything else?

[RFID.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11331021/RFID.zip)

LuciferShadows, 2023-04-28:
[image]

skotopes, 2023-05-15:
This doesn't looks like RFID or it requires reader to send request.
Do you have proxmark?

closed 2023-06-22 by skotopes