flipperdevices/flipperzero-firmware #2631: Suggestion: ability to take screenshots on actual flipper without phone connected, then save to SD card

Xbxjdjdjd, 2023-05-02:


skotopes, 2023-05-15:
Question is which key combo to use: all our keys are used in apps. We'll think about it

magopeter, 2023-05-23:
Other ideas to trigger a screenshot: 
Using a timer, like on a camera: by enabling this feature in a menu (ex. "Take screenshot in 15s") and then navigating to what you want to capture. The buzzer could also beep progressively faster until capture.
OR with GPIO pins. This might be harder to use and needs jumper wires, but GPIO pin 17 can be shorted via the iButton pogo pins with a penny ;)

skotopes, 2023-05-23:
That sounds quite crazy. Our UI/UX team is not going to be happy ;-)
I don't see how it can be implemented in user friendly manner so I'll close this ticket till someone propose acceptable solution.

closed 2023-05-23 by skotopes