flipperdevices/flipperzero-firmware #2634: Can't read my EGYM RFID card

Boringsapiens, 2023-05-04:
### Describe the enhancement you're suggesting.

Hello,

I tried to read my EGYM (Fitness club) RFID card in order to clone to the ring or bracelet but failed. 
[Fit123.psk.raw.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11398251/Fit123.psk.raw.zip)

Could you please analyze it and add it, if possible, for the upcoming releases? These EGYM machines are getting popular and I guess people will be interested in such capabilities.

Thanks!

skotopes, 2023-05-15:
You RAW file is damaged. Also they goes in a set of two.
Fitness clubs usually use NFC, are you sure that this card is LF RFID? Do you have proxmark?

Boringsapiens, 2023-05-16:
Sorry, my bad. Here are the new raw files and a couple of photos of the card. I tried to read the card with NFC readers - doesn't work. According to the vendor of EGYM it's an RFID card, but no specs if it's LF, HF or UHF. Unfortunately, I don't have the proxmark at hand but will try to get it. Also, will try to find info about the readers. Will keep you posted. 
https://egym.com/us/cloud/management-software
[Ffirst.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11484684/Ffirst.zip)
[image]
[image]

skotopes, 2023-05-16:
ASK RAW is empty, PSK doesn't contains any meaningful data. This is not LF RFID Card.

Boringsapiens, 2023-05-16:
So, it means that without the proxmark no options to check it, right? Will try to get it.

skotopes, 2023-05-17:
If flipper can not detect it then it's something exotic or complex. In that case PM is a last resort solution.

Boringsapiens, 2023-05-31:
Hi there, finally I got PM. It's Legic Prime MIM256. Are u interested in such type of cards to be added? I can provide a dump of it. Even two, if needed.

skotopes, 2023-06-06:
Yes, Legic prime eventually will be added. Most likely after we roll out new NFC architecture

Builderhummel, 2023-06-22:
This feature request is related [!1688 ](https://github.com/flipperdevices/flipperzero-firmware/issues/1688)

parsley, 2024-07-03:
Any news on Legic with Flipper?

skotopes, 2024-07-07:
@parsley nope, still in backlog