flipperdevices/flipperzero-firmware #2635: Weather Station read/edit/send new values

krisss117, 2023-05-04:
### Describe the enhancement you're suggesting.

Hi, it will be great to add ability to edit scanned parameters and send it to receiver.

skotopes, 2023-05-15:
Hi @krisss117, weather application is something we do in our free time. Saving, editing and transmitting clearly requires too much hustle with little to no use in real life.
However you are free to modify it.

closed 2023-05-15 by skotopes

Crsarmv7l, 2023-05-16:
@krisss117 I have a repo that can create various flipper weather signals (Lacrosse, Nexus-TH, bresser3ch). A major problem with trying to "spoof" a weather station, is that they typically update at low intervals (every 1 minute, 5 minutes, etc), you don't know so you would need to be transmitting the entire time.

I made this as a learning experience. It isn't very useful. 

https://github.com/Crsarmv7l/Python-Scripts-Flipper

Anything that can be decoded by rtl_433 can be recreated provided you have the time and knowledge to do so.