flipperdevices/flipperzero-firmware #2642: Shortcuts to change frequency in Sub-GHz Scan

xologram, 2023-05-06:
### Describe the enhancement you're suggesting.

Right now SubGHz Read places FZ in read mode with left key to go to config, leaving other buttons unused. Up/Down could be Freq+ and Freq- and right could cycle through modulation. I think it will make it more user friendly. Possible?

Skorpionm, 2023-05-09:
no, it will not be convenient, in fact, changing the frequency and modulation is not needed so often.  and by making them on hotkeys, the frequency and modulation will constantly go astray when worn in your pocket

xologram, 2023-05-09:
>changing the frequency and modulation is not needed so often.

i mean.. speak for yourself. given that you can listen the signal i would be useful to me if I could browse the frequencies while listening. 

the double click on the center button (or long press) could be used to "lock the hotkeys" (even the left key) to avoid the pocket bumps.

skotopes, 2023-05-15:
Kinda make sense, but we don't have enough buttons for it. I'll bring this topic to UI/UX team.

xologram, 2023-05-16:
If button hold can work than it would be great perfect:
Up/Down: Freq+/Freq-
Up hold/Down Hold: Modulation+/Modulation-
Left/Right: RSSI Threashold +/- (Like in Frequency Analizer)
Left Hold: Toggle hopping
Right Hold: Toggle Bin RAW
OK: Toggle Sound
OK hold: lock the shortcut buttons

But even without the hold I think it would be great to add just Freq+/Freq- and maybe Sound on/off. 

Thanks for looking into it.