flipperdevices/flipperzero-firmware #2652: Using random UID when emulating NFC cards

zhaorui2022, 2023-05-08:
### Description of the feature you're suggesting.

Hi,

What I am trying to do is emulating NFC cards/tags I own, however, during emulation, I want to use different UIDs every time I scan it. Some video games only allow scanning NFC tags with the same UID once per day, and I want to make it possible to scan multiple times by using random UIDs. Currently I see it is possible to create multiple cards with different UIDs, but it ends up with too many cards saved on the device, and the only difference is the UID. So if there are ways to use random UIDs during emulation, it will help a lot.

This might not be something everyone wants. I can fork this repo, make the change in the forked repo and load custom firmware on my device. However, I might need some help understand what files I should look at.

nvx, 2023-05-09:
Are you sure the tag data is independent of the UID though? This is the case for Amiibos for example and would require a bunch of extra work to make the game still read the card with a changed UID.

zhaorui2022, 2023-05-09:
I am actually referring to amiibo. The reason I have the impression that changing UID would work is that I watched a video of someone using .bin files directly with an emulator, the emulator supports generating random uuid, and then the person in the video was able to scan the same amiibo more than once a day.

skotopes, 2023-05-15:
Sounds like you are looking not for changeable UID but for generator that creates new cards. Clearly not something that we can put into firmware. But it's totally possible to make such things

closed 2023-05-15 by skotopes