flipperdevices/flipperzero-firmware #2654: Weather station app frequency hopping time!

schmiernippel, 2023-05-09:
### Describe the enhancement you're suggesting.

Weather stations are sending every 15-90sec a signal, so frequency hopping is to fast to catch the signals.
When you can set the hopping intervall freely, it would catch every station on every frequency.

I checked with cheap RTL-SDR my near stations, have added the frequencies to user frequencies in flipper, but hopping is to fast :-(

### Anything else?

Keep on your great work!!!

schmiernippel, 2023-05-09:
I see in the waterfall (sdrsharp) that every 30sec or 60sec or 80sec or max 90sec a beacon come from a station from at a unique frequency.

For example, weather app stays at every frequency in hopping list for 90sec...

A beacon come at 433.92 every 60sec.... F0 will catch it because F0 will stay 90sec in that frequency....

A beacon come at 868 every 30sec.... F0 will catch it because F0 will stay 90sec in that frequency....

A beacon come at 433.05 every 15sec.... F0 will catch it because F0 will stay 90sec in that frequency....

And so on.....

And then the list with received stations get longer and longer with time.....

Maybe i dont know what mean, but with rtl_433 i make this everytime!

Skorpionm, 2023-05-15:
the problem is that you do not know when 1 signal will arrive in order to synchronize on it. once every 60 sec. in fact, this time floats a little + - 0.5 sec, and this time is already more than enough to let the signal through

skotopes, 2023-05-15:
There are too many variants that we'll have to cover.

closed 2023-05-15 by skotopes