flipperdevices/flipperzero-firmware #2655: Device ID / Bluetooth MAC different on newer flipper

KJee85, 2023-05-09:
### Describe the bug.

In https://github.com/flipperdevices/flipperzero-firmware/blob/dev/firmware/targets/f7/furi_hal/furi_hal_version.c#LL110C49-L110C49 the BT MAC address is built from the device id from the stm32wb55 which says this should always be 0x26

On my flipper, my Bluetooth MAC is 0x27 for that part. Is this expected? I can't seem to find any docs for STM on why or what revision of the chip returns 0x27.

### Reproduction

1. Connect Flipper to BT, note MAC address like 80:E1:27...

hedger, 2023-05-09:
So, how is this a bug? What does it affect?

KJee85, 2023-05-09:
I'm looking at building some custom GPIO modules for my Flipper, and noticed the difference. and wanted to understand why so I can account for any differences in the chip.

hedger, 2023-05-09:
Those are different MCU revisions.

closed 2023-05-09 by hedger