flipperdevices/flipperzero-firmware #2663: saved infrared tx not generating correct carrier frequency

daviderud, 2023-05-11:
### Describe the bug.

The infrared raw transmission seem to not correctly handle the parameter "frequency:" (nor "duty cycle:").

Example with frequency:56000, a 37 KHz carrier is obtained instead

[image]

### Reproduction

1.Load a raw remote with frequency:56000. For example the attached one.

[ARRIS_test.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11448010/ARRIS_test.zip)

2.Activate the debug GPIO output and connect it to a scope.
3.Measure the subcarrier frequency

gsurkov, 2023-05-17:
Yep, the code outright ignores these values. Working on it.

skotopes, 2023-05-25:
Fixed in #2677, merged in dev and will be included in release 0.84

closed 2023-05-25 by skotopes

daviderud, 2023-05-28:
Confirmed working the with involved HW. Thanks!