flipperdevices/flipperzero-firmware #2664: "Forget All Paired devices devices" and "Bluetooth -> OFF" has no effect for Remote app

shchuko, 2023-05-11:
### Describe the bug.

Bluetooth remote keeps working after switching Bluetooth off and unpairing all the devices in the Bluetooth menu.

Firmware version is 0.82.3

### Reproduction

1. Turn on Bluetooth
2. Connect to your phone/laptop using Applications -> Bluetooth -> Remote
3. Exit the Remote app
4. Go to Settings -> Bluetooth, turn off Bluetooth and forget all paired devices
5. Go back to the Remote app and see that the remote keeps connected and working

An example is attached with screen recording (flipper bt remote is connected to a computer where qFlipper is opened, see the cursor behaviour)

https://github.com/flipperdevices/flipperzero-firmware/assets/36963534/dc060046-1b0f-4e82-9466-aee2e0815e1f

skotopes, 2023-05-15:
That is correct behavior. Remote application uses it's own set of keys and stores them on SD-Card.
We'll update remote application with unpair option.

hedger, 2024-01-10:
HID app now has "remove pairing" option.

closed 2024-01-10 by hedger