flipperdevices/flipperzero-firmware #2697: Requiring to guess modulation for sub-GHz seems silly

tko, 2023-05-24:
### Describe the enhancement you're suggesting.

The Frequency Analyzer seems to have no trouble recognizing signals / bursts regardless of modulation.

Having the Read function to _require_ the _user_ to correctly _guess_ the modulation seems very silly considering a computer should be able try all possibilities and pick the most likely one in microseconds without any user interaction.

I guess my suggestion is two-fold, independent of each other really:

1. have Frequency Analyzer show the most likely modulation for the signal it caught
2. have Read determine the modulation on its own

### Anything else?

I think whatever my key fob is sending isn't recognized by flipper, but rather than making me manually try every setting separately, the computer could try all the options in fraction of the time with no input from me and just tell me the signal is unknown.

skotopes, 2023-05-24:
Frequency Analyzer don't care about payload content only envelope, that is why it can detect signals in any modulation.
What you want requires SDR and multiple demodulators running in parallel and impossible to be implemented on Flipper Zero.

skotopes, 2023-05-24:
@tko basically radio works in quite predetermined way: antenna->amplifier->heterodyne->demodulation->decoding where all those steps are usually done in hardware.

Radio with 2 parallel demodulators are quite rare, so the thing that you want can be easily implemented in SDR(since demod step is in software), but can not be implemented on flipper(It uses CC1101 with fixed pipeline and demod done in chip itself).

tko, 2023-05-24:
Ah, thanks. Yes, too much SDR influence for me, no idea what CC1101 is. (yet)