flipperdevices/flipperzero-firmware #2703: Another unknow fob 125khz

gauthi3r, 2023-05-26:
### Description of the feature you're suggesting.

Hi,
i have one of these badge in 125khz, but my flipper can't read it : https://www.deister.com/uploads/tx_ttproducts/datasheet/tpx-keyfobs_datasheet_en_v160815.pdf 

Will we be able to read them soon with Hitag or Temic technology?

Thank,

skotopes, 2023-06-06:
Which one from this datasheet is yours?
There were talks about hitag, as for Temic we need more info and raw records.

gauthi3r, 2023-06-06:
The mine is TPX207 or 507 I don’t know exactly.
I send you raw file today.
Thx

g3kkorist, 2023-11-03:
Got m new Pet Yesterday and Check my 125khz fob, its a https://shop.deister.com/en/Keyfob-transponder-TPX207-OTP-125kHz-black/01588.000 and the little flipper wont read it. What can i do to help ?

Astrrra, 2024-01-11:
Closing this as inactive, please open a new issue if the problem is still present and you can provide the RAW files.

closed 2024-01-11 by Astrrra