flipperdevices/flipperzero-firmware #2704: Region not provisioned after update to latest official release firmware

alexminza, 2023-05-26:
### Describe the bug.

After updating to the latest official release firmware the provisioned region was lost and the Sub-GHz transmit on legal frequencies is blocked. Previously provisioned region was for Romania / EU.

```
[12:59:57][debug]: < response(system(deviceInfo(hardware_region: 4)))
[12:59:57][debug]: < response(system(deviceInfo(hardware_region_provisioned: --)))
```

[image]

[image]

[image]

FW Version Info: 0.83.1 [17-05-2023]

[image]

[image]

### Reproduction

Menu -> Sub-GHz -> Saved -> [Select saved item] -> Emulate -> Send

### Logs

```Text
[12:59:57][debug]: < response(system(deviceInfo(hardware_region: 4)))
[12:59:57][debug]: < response(system(deviceInfo(hardware_region_provisioned: --)))
```
```

tonyfreeman, 2023-05-26:
please report a bug trough the app and post issue id here

alexminza, 2023-05-26:
> please report a bug trough the app and post issue id here

How do you report a bug through the Flipper mobile app? Did not find any option for this...

tonyfreeman, 2023-05-26:
Options -> Report a Bug

alexminza, 2023-05-26:
> Options -> Report a Bug

No such option in the iOS app

[image]
[image]

tonyfreeman, 2023-05-26:
ouch, it's still in beta https://testflight.apple.com/join/oLyR7YeT

tonyfreeman, 2023-05-27:
if you don't want to install beta you could attach full log here, those two lines are not enough

tonyfreeman, 2023-05-27:
also please check does `/int/.region_data` exist?

closed 2023-05-29 by alexminza