flipperdevices/flipperzero-firmware #2706: Running `fbt vscode_dist` should ignore IDE files locally via `.git/info/exclude` rather than via `.gitignore`

AJMansfield, 2023-05-27:
### Describe the bug.

When running `fbt vscode_dist` in order to set up an _existing_ project for VSCode integration, `fbt` appends the IDE-specific ignores to `.gitignore`.

This is often the proper behavior when setting up preferred tooling for a project as a whole, but it's usually not appropriate to make changes specific to one's own tooling to a collaborative project -- even just a `.gitignore` line.

The correct way to locally ignore local IDE files is to instead append those lines to `.git/info/exclude`.

### Reproduction

```
$ git clone some@example.com:collaborative/project.git
$ cd project
$ fbt vscode_dist
$ git status
On branch master
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   .gitignore

no changes added to commit (use "git add" and/or "git commit -a")
```

### Target

fbt

skotopes, 2024-07-07:
@hedger ?

hedger, 2024-07-11:
@AJMansfield fbt is designed for running from firmware's source code root, not from anywhere else. And it comes with its own static .gitignoe, which it never modifies. I couldn't reproduce the issue using your instructions.

Maybe you meant ufbt and deploying app code template?

AJMansfield, 2024-07-16:
Looks like quite a lot has shifted in the toolchain since I last took a look at this, but I'm guessing you're right and this complaint is about ufbt now (if it's still current). The scenario I remember encountering this in was with setting up a project for a custom flipper app.

skotopes, 2024-07-22:
@AJMansfield can you test all your use cases with latest ufbt and sdk versions. Or it's ok to close this issues?

closed 2024-07-22 by AJMansfield