flipperdevices/flipperzero-firmware #2709: Can we know if we are reading a mifare plus card?

UrSuLa360, 2023-05-28:
### Description of the feature you're suggesting.

Im pretty sure im reading a mifare plus card but flipper says it is mifare classic.

I know mifare plus is compatible with mifare classic readers. 

But my question is: is it possible to detect when you are reading a mifare plus card type?

AloneLiberty, 2023-05-28:
Mifare Plus in SL1/SL2 can work as Mifare Classic, you are probably reading Mifare Plus in SL1. Currently isn't possible to separate Mifare Plus from any other tags with same SAK/ATQA (ATS required). This will be possible after NFC refactor (#2529).

skotopes, 2024-01-10:
@UrSuLa360 please check latest version. NFC refactoring changed way we detect card types in better and more reliable way.

UrSuLa360, 2024-01-10:
Hello, I no longer have any mifare plus card. I will try to get one.

UrSuLa360, 2024-01-21:
Got the card.
All interaction with the card was exactly like a Mifare Classic 1K.
[image]

First time i read the card the dict attack found 1 key. 
The rest of keys were found using _mfoc-hardnested_ on the computer.
Then i added all keys to user_dict in flipper and i was able to read the full content.

so, flipper could not detect the real type of the card.

One thing i want to point is, i know the card it's a Mifare Plus because the company using this card SAID it is a Mifare Plus SE.
Here is the link where they describe the card used: [https://blog.renfe.com/asi-funciona-la-tarjeta-sin-contacto/](url)
(It is in spanish)

Tomi2965, 2024-01-21:
Same problem here. Fob: FDI Matelec France Mifare Plus (part nr. GB-010-229). Readed on latest dev (027ea9ea).

[image]

closed 2024-07-07 by skotopes

ponty-pilot, 2025-02-12:
> Got the card. All interaction with the card was exactly like a Mifare Classic 1K. [image]
> 
> First time i read the card the dict attack found 1 key. The rest of keys were found using _mfoc-hardnested_ on the computer. Then i added all keys to user_dict in flipper and i was able to read the full content.
> 
> so, flipper could not detect the real type of the card.
> 
> One thing i want to point is, i know the card it's a Mifare Plus because the company using this card SAID it is a Mifare Plus SE. Here is the link where they describe the card used: [https://blog.renfe.com/asi-funciona-la-tarjeta-sin-contacto/](url) (It is in spanish)
> 
> Edit: tested on 0.97.1, my UID is 4 bytes long.

Please share the keys to the card. I want to try to write my train card to the flipper