flipperdevices/flipperzero-firmware #2718: Emulation timing regression after PR #2070

AloneLiberty, 2023-05-31:
### Describe the bug.

After merging PR #2070 frame delay time increased in 50%

In both cases Flipper has Debug: Off and Log Level: Default

### Reproduction

1. Emulate MFC on Flipper Zero
2. Read sector or block (hf mf rdsc -s 0 -k FFFFFFFFFFFF)
3. Check frame times (hf mf list --frame)
4. Revert commit e1c6e78b2ecdfe6da19c04d4e16b95c7f8247e9f and try again

### Logs

```Text
Trace before:

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52(7)                                                                    |     | WUPA
        992 |       2116 |     |fdt (Frame Delay Time): 1124
       2116 |       4484 | Tag |44  00                                                                   |     | 
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
       9504 |      10564 |     |fdt (Frame Delay Time): 1060
      10564 |      16452 | Tag |04  d3  c7  60  70                                                       |     | 
      19072 |      29600 | Rdr |93  70  04  d3  c7  60  70  74  b9                                       |  ok | SELECT_UID
      29600 |      30660 |     |fdt (Frame Delay Time): 1060
      30660 |      34180 | Tag |08  b6  dd                                                               |     | 
      36352 |      41056 | Rdr |60  00  f5  7b                                                           |  ok | AUTH-A(0)
      41056 |      47060 |     |fdt (Frame Delay Time): 6004
      47060 |      51796 | Tag |71  97  f8  ce                                                           |     | AUTH: nt 
      61184 |      70560 | Rdr |f2  2e  53! f5  cf  cc! 10  37                                           |     | AUTH: nr ar (enc)
      70560 |      78660 |     |fdt (Frame Delay Time): 8100
      78660 |      83396 | Tag |51! b4  e0  9f!                                                          |     | AUTH: at (enc)
      89344 |      94112 | Rdr |33  70! ae  fc!                                                          |     | 
            |            |  *  |                                              key FFFFFFFFFFFF prng HARD |     |
            |            |  *  |30  00  02  A8                                                           |  ok | READBLOCK(0)
      94112 |     113124 |     |fdt (Frame Delay Time): 19012
     113124 |     133988 | Tag |e8! fb! d4  e1  50  b3  6c  3e  06  d1  67! eb! 0a! 6a  d9! e9! 3b  69!  |     | 
            |            |  *  |04  D3  C7  60  70  08  44  00  FF  FF  FF  FF  FF  FF  FF  FF  EF  6F   |  ok | 
     147200 |     151968 | Rdr |c3  17! 93! 0c!                                                          |     | 
            |            |  *  |30  01  8B  B9                                                           |  ok | READBLOCK(1)
     151968 |     170884 |     |fdt (Frame Delay Time): 18916
     170884 |     191684 | Tag |cb! a5! de  95! 74! 98  12! c4  78! 32  6c! f8  e2! b4  d7  f5! c0! 51!  |     | 
            |            |  *  |FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  37  CB   |  ok | 
     204928 |     209632 | Rdr |13! 6a  51  ab!                                                          |     | 
            |            |  *  |30  02  10  8B                                                           |  ok | READBLOCK(2)
     209632 |     228564 |     |fdt (Frame Delay Time): 18932
     228564 |     249428 | Tag |cb  78! 90  e5! 87! f0  81! a2! b7! 0e  a0  7b  d3  f4! 3b  d2  2d  c6!  |     | 
            |            |  *  |FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  37  CB   |  ok | 
     262528 |     267232 | Rdr |5c  7b! 22! 0b!                                                          |     | 
            |            |  *  |30  03  99  9A                                                           |  ok | READBLOCK(3)
     267232 |     286308 |     |fdt (Frame Delay Time): 19076
     286308 |     307172 | Tag |60! 23! f0! 6b  a7! 88! 12! a6  8b! 8b  bf  fc  c9! 32! 1f! 83  9f! 70   |     | 
            |            |  *  |00  00  00  00  00  00  FF  07  80  69  FF  FF  FF  FF  FF  FF  D4  55   |  ok | 
     320256 |     325024 | Rdr |1b! 8b! 7a! 14!                                                          |     | 
            |            |  *  |50  00  57  CD                                                           |  ok | HALT

After:
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52(7)                                                                    |     | WUPA
        992 |       2116 |     |fdt (Frame Delay Time): 1124
       2116 |       4484 | Tag |44  00                                                                   |     | 
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
       9504 |      10564 |     |fdt (Frame Delay Time): 1060
      10564 |      16452 | Tag |04  e2  ff  09  10                                                       |     | 
      19072 |      29536 | Rdr |93  70  04  e2  ff  09  10  1a  78                                       |  ok | SELECT_UID
      29536 |      30660 |     |fdt (Frame Delay Time): 1124
      30660 |      34180 | Tag |08  b6  dd                                                               |     | 
      36352 |      41056 | Rdr |60  00  f5  7b                                                           |  ok | AUTH-A(0)
      41056 |      49348 |     |fdt (Frame Delay Time): 8292
      49348 |      54020 | Tag |df  de  45  96                                                           |     | AUTH: nt 
      63488 |      72800 | Rdr |9c  3b! 0b! b5! 9a  40  cf  31!                                          |     | AUTH: nr ar (enc)
      72800 |      83284 |     |fdt (Frame Delay Time): 10484
      83284 |      87956 | Tag |f9! 58  9d  37!                                                          |     | AUTH: at (enc)
      93952 |      98720 | Rdr |24  5c! 37! e9                                                           |     | 
            |            |  *  |                                              key FFFFFFFFFFFF prng HARD |     |
            |            |  *  |30  00  02  A8                                                           |  ok | READBLOCK(0)
      98720 |     127892 |     |fdt (Frame Delay Time): 29172
     127892 |     148692 | Tag |f8  c4  68  32  55! 3c! f4  5b  98! 5d  34! ab  34! 63! 4e! 08! 2f! 80!  |     | 
            |            |  *  |04  E2  FF  09  10  08  44  00  FF  FF  FF  FF  FF  FF  FF  FF  17  D6   |  ok | 
     161920 |     166688 | Rdr |33! 9e  33  bb!                                                          |     | 
            |            |  *  |30  01  8B  B9                                                           |  ok | READBLOCK(1)
     166688 |     195876 |     |fdt (Frame Delay Time): 29188
     195876 |     216740 | Tag |e7  97! 37! 53  81  e7  77  11  d3  20  98! ab  80  58! f1! 05  5b! 1e!  |     | 
            |            |  *  |FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  37  CB   |  ok | 
     229888 |     234592 | Rdr |d0! 0c  a9  7e                                                           |     | 
            |            |  *  |30  02  10  8B                                                           |  ok | READBLOCK(2)
     234592 |     263844 |     |fdt (Frame Delay Time): 29252
     263844 |     284708 | Tag |29  92! 1d  25! 13! e9! a9  f1! 33  9d! 11! c2! 09  95  7b  83  84  42!  |     | 
            |            |  *  |FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  FF  37  CB   |  ok | 
     297856 |     302560 | Rdr |68! 39  af  02!                                                          |     | 
            |            |  *  |30  03  99  9A                                                           |  ok | READBLOCK(3)
     302560 |     331796 |     |fdt (Frame Delay Time): 29236
     331796 |     352660 | Tag |3a  d6  3f! d9! 1c! cc! 50  c5  e9! ae  c6  9f! a8! d5! 4b  fd! 70  c0!  |     | 
            |            |  *  |00  00  00  00  00  00  FF  07  80  69  FF  FF  FF  FF  FF  FF  D4  55   |  ok | 
     365696 |     370400 | Rdr |d5! d2  0b  fb!                                                          |     | 
            |            |  *  |50  00  57  CD                                                           |  ok | HALT
```

closed 2023-06-01 by skotopes

g3gg0, 2023-06-02:
using the modification I have [undone in this patch](https://github.com/flipperdevices/flipperzero-firmware/commit/a1ae1ed515293b33d87d845acf6cbe5d898b6659#diff-183eae0deedddf39290d7a1bce14e5ddc0654de127eea508df4dc20d358b7b63), i get these timings:

[image]

however, this was done using the assembly optimized variant of the DigitalSequence which was faster.
having some issues with the current implementation. trying to nail them down.

AloneLiberty, 2023-06-02:
> however, this was designed for the assembly optimized variant of the DigitalSequence which was faster.

Good job, can we assume this is ideal Flipper emulation and it can't get any faster (not by that much)?

g3gg0, 2023-06-03:
this is not ideal by any means. this is a tradeoff between pre-calculating all pin levels and calculting them while already sending them.
there is still the protocol handling stuff happening before, decoding the request as well as encoding the response.
you can loose and win cpu cycles at any step...

right now i am struggling in making DigitalSequence faster, running into some race conditions.
so do not expect a quick update from my side soon, and if, it would take some time for testing it.

PR #2719 (2023-06-01): NFC: fix MFC timings
# What's new

- Optimize DigitalSignal lib to improve MFC emulation
- Build DigitalSignal as separate library
- Improve unit tests
- Fixes #2718 

# Verification 

- During MFC emulation Nt response fits in 500us timeout
- Confirm NFC unit tests don't fail

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix