flipperdevices/flipperzero-firmware #2722: More than one Flipper is attached

johmarjac, 2023-06-01:
Hi,

I just recursive cloned the repository on my MacBook Air M1, invoked `./fbt vscode_dist` to setup the vs code configuration (which unfortunately didn't fix include errors but that's not my problem rn), I copied the Example Thermo into applications_user folder, emptied it out so it really just is a baritone app and it compiles fine. But when I use command of VS Code to launch app on Flipper it says in console:

`2023-06-01 21:09:34,410 [ERROR] More than one Flipper is attached
2023-06-01 21:09:34,410 [ERROR] Failed to guess which port to use`

Can I somehow list my devices and maybe invoke a parameter to tell it which one to use? Although I only have one Flipper attached via USB lol.

hedger, 2023-06-01:
Can you please show the output of `ls /dev/cu*` ?

hedger, 2023-06-01:
Oh I see. `JBLFlip5` is confusing Flipper detection code, since it uses the same prefix as Flipper's virtual COM ports.

Let me think of a workaround.

johmarjac, 2023-06-01:
> Oh I see. `JBLFlip5` is confusing Flipper detection code, since it uses the same prefix as Flipper's virtual COM ports.
> 
> Let me think of a workaround.

Awesome thanks :D

hedger, 2023-06-01:
Well, a quick and dirty fix is:
https://github.com/flipperdevices/flipperzero-firmware/blob/dev/scripts/flipper/utils/cdc.py#L9
Change "flip" -> "flip_"

johmarjac, 2023-06-01:
> Well, a quick and dirty fix is:
> https://github.com/flipperdevices/flipperzero-firmware/blob/dev/scripts/flipper/utils/cdc.py#L9
> Change "flip" -> "flip_"

Yes this worked, Thanks now I can continue.

PR #2723 (2023-06-08): [FL-3346] fbt: added Flipper selection when multiple are connected over USB
# What's new

- [FL-3346] fbt: added Flipper selection when multiple are connected over USB
- Also fixes #2722

# Verification 

- Connect more than one fiipper over USB
- Check that you can select which one to use for `fbt cli`, `fbt flash_usb`, etc by adding `FLIP_PORT=/dev/cu....`

# Checklist (For Reviewer)

- [ ] PR has description of feature/bug or link to Confluence/Jira task
- [ ] Description contains actions to verify feature/bugfix
- [ ] I've built this code, uploaded it to the device and verified feature/bug [... cut]
closed 2023-06-08 by skotopes