flipperdevices/flipperzero-firmware #2727: Invalid Troika balance parsing for sublayout 0x5

monster1025, 2023-06-03:
### Describe the bug.

Flipper supports now only troika with old (outdated) layout 0x3, but currently used layout is 0x5.
The displayed balance is wrong, for example for my card with 3194rub balance flipper displays 572rub.

Here is description for outdated layout 0x3:
https://github.com/metrodroid/metrodroid/wiki/Troika#e-pursebalance-sector-8-layout-0xe-sublayout-0x3

Here is sample data and script for layout 0x3 and 0x5:
https://github.com/metrodroid/metrodroid/issues/735

In layout 0x3 balance in bits 188:209 ( from sector start,  length 22).
In layout 0x5 balance in bits 165:185 ( from sector start,  length 20).

# troika - new format (layout E/subLayout 5), balance 348 (rub), whole sector 8
data = '45DB101958FBCE2A4915216AA1A008000D1C0A000004E20B004001001EB7ADDE0D1C0A000004E20B004001001EB7ADDE'

# troika - old format (layout E/subLayout 3), balance 0 (rub).
data = '45DB101958FBCE19768AA400000000002C013D460A001400000010009BB56E632C013D460A001400000010009BB56E63'

### Reproduction

1. Switch on
2. Open menu, go to NFC - Read
3. Read Troika card
4. Compare displayed balance by flipper and official metro android app.

### Target

Flipper + Troika

doomwastaken, 2023-11-03:
Issue still persists, couldn't find a task for it. Will update once fixes are in dev tree

doomwastaken, 2023-11-21:
Currently in work, ticket (FL-3661)

skotopes, 2024-04-14:
Should be fixed in latest release.

closed 2024-04-14 by skotopes