flipperdevices/flipperzero-firmware #2734: Mark already saved tags after scanning

lifeart, 2023-06-05:
### Description of the feature you're suggesting.

Once `rfid`/`nfc`/`iButton` item is scanned, and it's **already saved** in flipper (known) - show Icon / or quick action to go to saved item.

Problem: It's really easy to forget if item is already scanned / saved (if we have few marks with same color / shape)
Solution: Eliminate double save and allow user go to saved mark to view / edit.

Razberry314, 2023-06-06:
I think this is good, but you should still be able to save it twice for things such as if you want one to be written to, and one to stay the same, or having them in different folders. Perhaps it would do a little "warning" that would tell you it's already saved?

skotopes, 2023-06-06:
Already was asked before: computationally expensive, going to be slow and complex to implement.

closed 2023-06-12 by hedger