flipperdevices/flipperzero-firmware #2749: Broken write support to iButton DS1971

MichaelGrafnetter, 2023-06-08:
### Describe the bug.

The iButton Write Copy feature does not work properly with DS1971. Instead of the actual data, the iButton end up containing value 0xFF 32 times (256 ones in binary).

### Reproduction

1. Open the iButton menu.
2. Select "Read" and scan an existing DS1971 key.
3. Select "More"
4. Navigate to "Write Copy".
5. Write a different (or same) DS1971 key.
6. Select "Back", "Retry", "Exit".
7. Scan the iButton from step 5.
8. The 256-bit Internal EEPROM now contains only FF FF...FF instead of the expected data.

MichaelGrafnetter, 2023-06-13:
Tried it again (both with 0.84 and 0.85 RC firmware), without any luck.

Trying to write this file:
```
Filetype: Flipper iButton key
Version: 2
Protocol: DS1971
Rom Data: 14 75 3A 07 09 00 00 4D
Eeprom Data: 50 68 <REDACTED...> 85 07 77 95 E0
```

Target is the same iButton, from which the data was originally fetched:

[image]

After I try to read the data from the iButton again after writing it, I always the same result:

[image]

The iButton is thus effectively bricked, as the entry system does not check the ID, but the EEPROM data.

Not sure how to troubleshoot this further.

doomwastaken, 2023-06-13:
[image]
I see, let me try and find your version of the ibutton

MichaelGrafnetter, 2023-07-29:
Tried it after a while with the latest FW (0.87.0) and it is now working without any issues. Which is strange, as I do not see any relevant changes in the repo.

MichaelGrafnetter, 2023-08-05:
I did some more testing and the success rate of writing data to DS1971 EEPROM is around 1:4. I wonder if Flipper could try to validate the data and re-write them in case of a mismatch.

skotopes, 2024-01-10:
@doomwastaken any progress?

doomwastaken, 2024-01-12:
I had re-tested it again on 1971 original and noname keys, did 10 writes with each of individual data sets, all seems to work correctly. I will close the issue. 

If you could provide us with information where to purchase same 1971 key that you are using, then I will order some and re-test this again.

closed 2024-01-12 by doomwastaken