flipperdevices/flipperzero-firmware #2770: [NFC Magic] Add support for backdoor read commands for Gen 1A/1B tags

mishamyte, 2023-06-13:
### Description of the feature you're suggesting.

**Overview:**

Right now NFC Magic app supports next functionality for magic tags:
* Detection
* Writing
* Wiping

It would be nice to add also support of read backdoor command. For that case flow of reading could be simplified. Instead of trying to authenticate with user dictionary + standard dictionary, tag could be read fully. 

**Usage steps:**
1. Check if tag is magic Gen 1A/1B (existing functionality)
2. Select read (for implementation)
3. If tag was read without problems (potential problems while reading should be an extra investigation points), user could have a possibility to save the dump (and next steps should be done using standard Flipper's NFC experience)
4. _[Optional]:_ As a follow up for step 3, found keys could be added into user's dictionary (automatically or with user confirmation, validation for key existence in dictionary should be done)

**Tech details for read commands**:
* _Gen 1A:_  `40(7)`, `43`, `30xx` + crc
* _Gen 1B:_  `40(7)`, `30xx` + crc (need to be validated should crc be present here)

**Potential problems**:
Now detection of Gen 1 tags is done, using `40(7)` command and concrete type (Gen1A/Gen1B) is not detected. For read it could be done via iteration through the read sequence for Gen1A, if didn't worked - for Gen1B. Which could ruin UX from the PoV of waiting time.

**Implementation in existing products**:
It is implemented at least in next solutions:
* Proxmark3 - `cview` command
* TMD-5S - it is used by default, if magic tag is detected

sealldeveloper, 2023-10-05:
+1 on this, would really like to see this as a feature.

spetzreborn, 2024-04-13:
I also want this feature. I have the possibility to use a card printer/writer. And this way would make it very easy for me to get the B keys as I just can write/print to an gen1a magic card.

Maybe this issue should be on the apps repo and not firmware as nfc magic is an standalone app there?

skotopes, 2024-04-14:
Latest version NfcMagic support gen1/gen2 writing.

mishamyte, 2024-04-14:
@skotopes I'm sorry, but the issue was about the backdoor _**read**_, which is still now implemented.

I agree that it would be better to have it in good faps repo. 

Would it be good if I will reopen it there as a feature proposal?