flipperdevices/flipperzero-firmware #2775: Mifare Plus golmar key is not emulated properly

mgrybyk, 2023-06-13:
### Describe the bug.

Emulation of golmar 13.56MHz tag that supports anti-copy encryption AES-128 doesn't work correctly.

### Reproduction

1. Read [golmar](https://www.golmar-seguridad.es/productos/tagdoor-mf-plus#product) key
2. Emulate it

**Actual Result**
the door won't open

**Expected Result**
the door should open

**Additional Info**
- my key [key.nfc.txt](https://github.com/flipperdevices/flipperzero-firmware/files/11729188/key.nfc.txt)
- I have 6 different readers, tried to use detect reader feature for all of them but still can see only sector 0. See [mfkey32.log](https://github.com/flipperdevices/flipperzero-firmware/files/11740128/mfkey32.log).
- the code written on the key is `80555B62663004`

### Logs

```Text
Filetype: Flipper NFC keys
Version: 1
Mifare Classic type: 1K
Key A map: 000000000000FFFF
Key B map: 000000000000FFFF
Key A sector 0: 88 29 DA 9D AF 76
Key B sector 0: 88 29 DA 9D AF 76
```

### Anything else?

cc @Astrrra

Astrrra, 2023-06-14:
So, from looking at the dump and the detect reader log, everything _seems_ fine. The only issue I can think of with this is that the reader sends a RATS command (a Mifare Plus exclusive command that isn't supported by Mifare Classic), and Flipper doesn't reply (because it only supports Mifare Classic cards, Mifare Plus cards aren't supported). Still, without a proxmark3 trace of the communication, I can't be sure about this.

Your Mifare Plus card is in SL1 (Security Level 1) which makes it act mostly like a Mifare Classic card (with Crypto1 instead of AES, btw), and Flipper detects it as Mifare Classic, and treats it as such, ignoring the Mifare Plus-specific features of the card (as it's not aware of them at all).

Basic Mifare Plus support is probably coming after the NFC refactoring (in a few months), and it will likely solve your issue (it will not, however, give Flipper the ability to read cards in SL3 with keys that you don't know, AES is still AES). For now, without a trace of the communication - that's all I can say.

mgrybyk, 2023-06-14:
@Astrrra thanks for the answer.
I don't have proxmark3. Can I somehow collect more logs with the flipper to help with investigation?

AloneLiberty, 2023-07-14:
There are some fixes with RATS on Flipper Zero emulation in dev firmware, I suggest you trying it.

mgrybyk, 2023-07-15:
@AloneLiberty I have to say that something has changed indeed.
Previously, the key reader didn't react anyhow on attempt to emulate the saved key, while it blinks with the red color now saying that the key is invalid.
Also, I can't get anything from the key reader now using the Detect Reader feature. 
Is there anything I can do to assist?
Thank you!

AloneLiberty, 2023-07-15:
Can you show trace logs while emulation and detect reader? https://play.google.com/store/apps/details?id=jp.sugnakys.usbserialconsole run `log trace`

mgrybyk, 2023-07-16:
@AloneLiberty I've attached the logs requested: 
[serial_20230716_224224.txt](https://github.com/flipperdevices/flipperzero-firmware/files/12065052/serial_20230716_224224.txt)

AloneLiberty, 2023-07-17:
Checked your logs, they looks very suspicious for me. REQA/WUPA somehow leaking into emulator (should be controlled by the NFC Chip?)
Next, "Error in tx rx" means that auth is succesful, Tx: 144 bits -> 18 bytes, should be block reading. Error seems to go into rfalIsExtFieldOn (all other cases should log timeout error).
I have some thoughts what could be a problem here. Could you contact me at Telegram (t.me/libertydev)? I will have a look when I will have some free time.

doomwastaken, 2023-11-03:
@mgrybyk @AloneLiberty got any new details? Also could you, please, check the issue on latest firmware

mgrybyk, 2023-11-03:
I'll try in the middle if next week.

mgrybyk, 2023-11-11:
@doomwastaken @AloneLiberty  I've tested the latest dev firmware build 49dcf817.
The flipper device won't react to the reader anyhow now. 
Detect reader does nothing, emulating the key does nothing. 
There are no log entries related to nfc emulation or detect reader.

skotopes, 2024-01-10:
@gornekich FYI

mgrybyk, 2024-01-10:
Upd from my side. As for now, it doesn't work like it didn't work before.
I mean the flipper reacts to flipper but the door won't open.

skotopes, 2024-07-07:
@mgrybyk how about now?

mgrybyk, 2024-07-07:
@skotopes behavior seems to be the same: the flipper reacts to flipper but the door won't open.

0.103.1

skotopes, 2024-07-07:
@mgrybyk

- What type of card do you have? Is it mifare plus? When you reading it with latest firmware what do you see?
- What type of reader do you have? Is it trying to read mifare card in classic or plus mode?
- Logs from flipper
- Full reder-card exchange with proxmark