flipperdevices/flipperzero-firmware #2777: RFID Raw - Keri Systems PKT-10X Standard Light Proximity Key Tag

JReming85, 2023-06-16:
### Describe the enhancement you're suggesting.

Would love it if you could implement this RFID tag.

They are the key tag our area uses to allow us to get into the mail and laundry room and pool.

We only get 1, and if we lose it we have to pay $50 for a replacement.  Would rather just clone it and keep the original safe.
[rfid backups.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11764822/rfid.backups.zip)

skotopes, 2023-06-22:
Some keri tags are already supported.
Raw files looks a little bit strange, looks like we need PKT-10X at our hands to implement them.

Also do you have proxmark?

JReming85, 2023-06-22:
Sorry I do not.  I can rescan raws of the tag if that will help

JReming85, 2023-06-22:
Here are 2 additional raw scans I just did away from all my other equipment.  Hopefully it helps

[raw.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11833593/raw.zip)

skotopes, 2023-06-27:
Yep, it may be rfid, but coding is quite unusual. We'll try to implement it

jbeoris, 2024-09-24:
Any status on this? @skotopes @doomwastaken ? I have the same issue.

skotopes, 2024-09-24:
@jbeoris in backlog waiting for someone to implement it

forxhunter, 2024-12-09:
please support this !!!

JJazz-jpg, 2025-03-19:
Interested as well

Ravioli-cpu, 2025-09-03:
@skotopes @doomwastaken hello, was wondering if there were any updates? Thanks so much