flipperdevices/flipperzero-firmware #2781: ISO 15693 SLIX2 signature support

eychei, 2023-06-16:
### Description of the feature you're suggesting.

The SLIX2 emulation is incomplete. 
The following feature is missing. 

• Originality signature:
32 byte ECC based originality signature

g3gg0, 2023-06-18:
Seems there is already a [tool for cracking the private key](https://github.com/RfidResearchGroup/proxmark3/blob/master/tools/recover_pk.py)...
[image]

so the signature might be possible to autogenerate for any UID.

for now i implemented all the emulation stuff for having a static signature, optional passwords etc.
unfortunately reading the signature doesn't work. the proxmark aborts the field in middle of the communication.
will have to dig deeper. do not know if this is the proxmark acting weird or the flipper having trouble sending longer responses...

[image]

g3gg0, 2023-06-18:
i wonder if the pm3 works with the SLIX2 as intended.
do you have a proxmark for testing?

[image]

eychei, 2023-06-18:
Great work ! I do have a proxmark  and can test tomorrow.

Frostyy99, 2023-06-18:
How do I add this to my flipper zero

eychei, 2023-06-18:
How does the private key cracking work here ? Are the signatures generated with weak nonces? how could you crack the private key with just one signature ?
We would need more then 50-60 biased signatures for a lattice attack. 

-e

g3gg0, 2023-06-18:
I have no clue, didnt expect that t.b.h.

However, from what i understand, it is just the public key used to verify the signature.
Recovering the private key is probably pointless.
ECC is a lot more complex than e.g. RSA, so even just 256 bit is a hard nut to crack.

The code is not live yet. I first want to fix the error when responding the 32 byte signature.
For this i have to know if pm3 works reliable.

eychei, 2023-06-18:
Trying to emulate the SLIX2 with the proxmark right now.
There is only a "sim" mode which only uses the UID.
Is there an emulation of ISO15693 SLIX2 Tags which I am missing?

I am on Iceman/master/v4.14831-643-ga0ac40449.
I only have the 256kb version.

-e

g3gg0, 2023-06-18:
there is no full emulation, I just implemented SLIX-L stuff a while back.
I don't think someone pushed that further since then.

can you read the SLIX2-tag? using "hf 15 info"

eychei, 2023-06-18:
Yes I can read the slix2 tags without issues. It also reads out the Signature:
Looks like this:

[+]  UID: E0 04 01 08 2F 81 D8 FC
[+] TYPE: NXP(Philips); IC SL2 ICS20/ICS21(SLI) ICS2002/ICS2102(SLIX) ICS2602(SLIX2)
[+] Using UID... E0 04 01 08 2F 81 D8 FC

[=] --- Tag Information ---------------------------
[=] -------------------------------------------------------------
[+]       TYPE: NXP(Philips); IC SL2 ICS20/ICS21(SLI) ICS2002/ICS2102(SLIX) ICS2602(SLIX2)
[+]        UID: E0 04 01 08 2F 81 D8 FC
[+]    SYSINFO: 00 0F FC D8 81 2F 08 01 04 E0 01 3D 4F 03 01
[+]      - DSFID supported        [0x01]
[+]      - AFI   supported        [0x3D]
[+]      - IC reference supported [0x01]
[+]      - Tag provides info on memory layout (vendor dependent)
[+]            4 (or 3) bytes/blocks x 80 blocks
[=] --------- NXP Sysinfo ---------
[=]   raw : 00 32 02 0F 7F 35 00 00
[=]     Password protection configuration:
[=]       * Page L read not password protected
[=]       * Page L write password protected
[=]       * Page H read not password protected
[=]       * Page H write not password protected
[=]     Lock bits:
[=]       * AFI locked
[=]       * EAS locked
[=]       * DSFID locked
[=]       * Password protection configuration locked
[=]     Features:
[=]       * User memory password protection supported
[=]       * Counter feature supported
[=]       * EAS ID supported by EAS ALARM command
[=]       * EAS password protection supported
[=]       * AFI password protection supported
[=]       * Extended mode supported by INVENTORY READ command
[=]       * EAS selection supported by extended mode in INVENTORY READ command
[=]       * READ SIGNATURE command supported
[=]       * Password protection for READ SIGNATURE command not supported
[=]       * STAY QUIET PERSISTENT command supported
[=]       * ENABLE PRIVACY command supported
[=]       * DESTROY command supported
[=]       * Additional 32 bits feature flags are not transmitted

[=]   EAS (Electronic Article Surveillance) is not active

[=] --- Tag Signature
[=]     Elliptic curve parameters: NID_secp128r1
[=]              TAG IC Signature: 22B83A239BE578F02CDED3CD31BA6A20F19AE721F2FCCCEE37FA68E77A298620
[+]        Signature verification: failed

g3gg0, 2023-06-18:
okay then sending responses >32 byte causes problems with the current NfcV implementation

g3gg0, 2023-06-18:
[image]

g3gg0, 2023-06-19:
[SKU_S0722550 S0722430.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11782907/SKU_S0722550.S0722430.zip)

@eychei can you test the linked PR if it works for you?

eychei, 2023-06-19:
Nice! Will try and report back.

eychei, 2023-06-19:
Ok I tried. Do get some errors / problems:

1. Proxmark3 does not recognize the tag immidaitely. I do have to search for the tag (emulated) a few dozen times till it gets recognized
2. On the original reader (Dymo) it does not get past the inventory command. The debug. txt does only contain these commands: 
1035227 R: 36 01 00 00 6a a1
1035228 R: 36 01 00 00 6a a1
1035228 R: 36 01 00 00 6a a1
1035228 R: 36 01 00 00 6a a1
1035244 R: 36 01 00 00 6a a1
1035247 R: 36 01 00 00 6a a1
1035247 R: 36 01 00 00 6a a1

I think this may be a timing issue?

-e

g3gg0, 2023-06-19:
damn.
did you try to disable debug logging to reduce latency?

g3gg0, 2023-06-19:
>     1. Proxmark3 does not recognize the tag immediately. I do have to search for the tag (emulated) a few dozen times till it gets recognized

yeah, seen that too. 
the first reading goes through, then the first few seconds it doesnt respond.

g3gg0, 2023-06-19:
yeah, also seen just then :D 
can you try the latest changes?

the behavior of the optional AFI field was implemented incorrectly

g3gg0, 2023-06-19:
looks good. can you sniff what an original tag responds?

eychei, 2023-06-19:
Ok got the proxmark log. Hope it is complete:

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |      52224 | Rdr |36  01  00  00  6a  a1                                                   |  ok |
 INVENTORY
    9636608 |    9688832 | Rdr |36  01  00  00  6a  a1                                                   |  ok |
 INVENTORY
    9692992 |    9746240 | Tag |00  01  fc  d8  81  2f  08  01  04  e0  cc  48                           |  ok |

    9795712 |    9913472 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  00  03  39  f0                   |  ok |
 READ_MULTI_BLOCK(0-3)
    9917632 |    9999552 | Tag |00  03  0a  82  ed  86  39  61  d2  03  14  1e  32  b6  ca  00  3c  d4   |     |

            |            |     |c3                                                                       |  ok |

   10096768 |   10140800 | Rdr |22  bd  e4  53  36                                                       | !crc|
 READ_SIGNATURE
   10144960 |   10288320 | Tag |00  22  38  23  9b  e5  78  f0  2c  de  d3  cd  31  ba  6a  20  f1  9a   |     |

            |            |     |e7  21  f2  fc  cc  ee  37  fa  68  e7  7a  29  86  20  67  cc           | !crc|

   11964864 |   12066240 | Rdr |22  2b  fc  d8  81  2f  08  01  04  e0  3e  af                           |  ok |
 GET_SYSTEM_INFO
   12070400 |   12144128 | Tag |00  0f  fc  d8  81  2f  08  01  04  e0  01  3d  4f  03  01  38  5a       |  ok |

   12338784 |   12456544 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  00  03  39  f0                   |  ok |
 READ_MULTI_BLOCK(0-3)
   12460736 |   12542656 | Tag |00  03  0a  82  ed  86  39  61  d2  03  14  1e  32  b6  ca  00  3c  d4   |     |

            |            |     |c3                                                                       |  ok |

   12604000 |   12721760 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  03  0f  3d  10                   |  ok |
 READ_MULTI_BLOCK(3-18)
   12970624 |   13080192 | Rdr |22  23  fc  80  2f  08  01  04  e0  14  07  ec  44                       | !crc|
 READ_MULTI_BLOCK(7-243)
   13084352 |   13231808 | Tag |00  d7  fa  00  1c  76  c6  91  7b  00  30  30  30  30  30  30  30  30   |     |

            |            |     |30  30  00  00  00  f9  0c  10  00  00  00  00  00  00  00  03  28       |  ok |

   13311968 |   13429728 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  1e  0c  4f  07                   |  ok |
 READ_MULTI_BLOCK(30-42)
   13690752 |   13808512 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  32  0b  63  f9                   |  ok |
 READ_MULTI_BLOCK(50-61)
   14209088 |   14220352 | Rdr |22                                                                       |     |

   14635904 |   14652288 | Tag |00  78  f0                                                               |  ok |

   14913056 |   15030816 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  4f  01  85  16                   |  ok |
 READ_MULTI_BLOCK(79-80)
   15035008 |   15067776 | Tag |00  fd  ff  00  01  a9  e4                                               |  ok |

   36301760 |   36337600 | Rdr |22  26  fc  d8                                                           | !crc|
 RESET_TO_READY
   36341312 |   36357696 | Tag |00  78  f0                                                               |  ok |

   49795456 |   49896832 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
   49901056 |   49917440 | Tag |00  78  f0                                                               |  ok |

   63333664 |   63484192 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2  00  00  00  00  00  40   | !crc|
 RESET_TO_READY
   72993824 |   73095200 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
   73099392 |   73115776 | Tag |00  78  f0                                                               |  ok |

g3gg0, 2023-06-19:
great, what does the log say for FZ?

eychei, 2023-06-19:
FZ log:

2912674 R: 36 01 00 00 6a a1
2912675 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
2912675 R: 36 01 00 00 6a a1
2912675 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
2912676 R: 36 01 00 00 6a a1
2912676 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
2912677 R: 36 01 00 00 6a a1
2912677 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
2912697 R: 36 01 00 00 6a a1
2912697 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48

g3gg0, 2023-06-19:
no, i mean proxmark log :)

eychei, 2023-06-19:
Somehow I did get some more transactions now with the dymo reader + Flipper:
```
   90015296 |   90067520 | Rdr |36  01  00  00  6a  a1                                                   |  ok |
 INVENTORY
   90071744 |   90124992 | Tag |00  01  fc  d8  81  2f  08  01  04  e0  cc  48                           |  ok |

   90207296 |   90259520 | Rdr |22  23  fc  d8  81  cb                                                   | !crc|
 READ_MULTI_BLOCK(252-468)
   92314816 |   92432576 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  03  0f  3d  10                   |  ok |
 READ_MULTI_BLOCK(3-18)
   93046752 |   93098976 | Rdr |22  23  fc  d8  81  07                                                   | !crc|
 READ_MULTI_BLOCK(252-468)
   93765824 |   93875392 | Rdr |22  b2  04  fc  d8  81  2f  08  01  04  e0  8f  9b                       |  ok |
 GET_RANDOM_NUMBER
   93883712 |   93908288 | Tag |00  e1  71  83  54                                                       |  ok |

   94245248 |   94330240 | Rdr |22  b3  04  fc  d8  81  2f  08  01  72                                   | !crc|
 SET_PASSWORD
   94523936 |   94641696 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  4f  01  85  16                   |  ok |
 READ_MULTI_BLOCK(79-80)
   94645888 |   94666368 | Tag |01  0f  68  ee                                                           |  ok |

  102097472 |  102403648 | Rdr |36  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00   |     |

            |            |     |00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00   |     |

            |            |     |00                                                                       | !crc|
 ?
  102540320 |  102649888 | Rdr |22  bd  04  fc  d8  81  2f  08  01  04  e0  53  36                       |  ok |
 READ_SIGNATURE
  103914496 |  104015872 | Rdr |22  2b  fc  d8  81  2f  08  01  04  e0  3e  af                           |  ok |
 GET_SYSTEM_INFO
  104020032 |  104093760 | Tag |00  0f  fc  d8  81  2f  08  01  04  e0  01  3d  4f  03  01  38  5a       |  ok |

  104288416 |  104406176 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  00  03  39  f0                   |  ok |
 READ_MULTI_BLOCK(0-3)
  104410624 |  104492544 | Tag |00  03  0a  82  ed  86  39  61  d2  03  14  1e  32  b6  ca  00  3c  d4   |     |

            |            |     |c3                                                                       |  ok |

  105808256 |  105926016 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  32  0b  63  f9                   |  ok |
 READ_MULTI_BLOCK(50-61)
  106275936 |  106385504 | Rdr |22  b2  04  fc  d8  81  2f  08  01  04  e0  8f  9b                       |  ok |
 GET_RANDOM_NUMBER
  122049120 |  122150496 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  122154688 |  122171072 | Tag |00  78  f0                                                               |  ok |

  135641696 |  135677536 | Rdr |22  26  fc  d8                                                           | !crc|
 RESET_TO_READY
  135681664 |  135698048 | Tag |00  78  f0                                                               |  ok |

  149133312 |  149234688 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  149238848 |  149255232 | Tag |00  78  f0                                                               |  ok |

  162692992 |  162794368 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  172317696 |  172419072 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  172423232 |  172439616 | Tag |00  78  f0                                                               |  ok |

  185877280 |  185978656 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  185982784 |  185999168 | Tag |00  78  f0                                                               |  ok |

  199404192 |  199505568 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  199509696 |  199526080 | Tag |00  78  f0                                                               |  ok |

[... 3 lines cut ...]

  226490912 |  226592288 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  226596480 |  226612864 | Tag |00  78  f0                                                               |  ok |

  240050528 |  240151904 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  240156032 |  240172416 | Tag |00  78  f0                                                               |  ok |

  253577536 |  253678912 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  253683072 |  253699456 | Tag |00  78  f0                                                               |  ok |

  267137216 |  267238592 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  267242752 |  267259136 | Tag |00  78  f0                                                               |  ok |

  280664128 |  280765504 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  280769664 |  280786048 | Tag |00  78  f0                                                               |  ok |

  294223840 |  294325216 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  294329344 |  294345728 | Tag |00  78  f0                                                               |  ok |
```

eychei, 2023-06-19:
I also can see that the flipper does have some data in the log.
Here is the log for the SKU_S0722430.nfc:

```
7478244 R: 36 01 00 00 6a a1
7478244 T: 00 01 ba 6c 60 3d 08 01 04 e0 5d 2b
7478244 R: 36 01 00 00 6a a1
7478245 T: 00 01 ba 6c 60 3d 08 01 04 e0 5d 2b
7478245 R: 22 23 ba 6c 60 3d 08 01 04 e0 00 03 21 a1
7478246 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
7478246 R: 22 bd 04 ba 6c 60 3d 08 01 04 e0 c2 55
7478247 T: 00 33 4a 63 63 d0 13 49 db a0 9e ee 15 1e f8 f8 f3 fa 15 f5 77 e4 4d 75 9b 78 14 ca d3 7e 02 ef 10 6d 93
7478248 R: 22 2b ba 6c 60 3d 08 01 04 e0 af cc
7478249 T: 00 0f ba 6c 60 3d 08 01 04 e0 01 3d 4f 03 01 40 d3
7478249 R: 22 23 ba 6c 60 3d 08 01 04 e0 00 03 21 a1
7478250 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
7478250 R: 22 23 ba 6c 60 3d 08 01 04 e0 03 0f 25 41
7478251 T: 00 b6 ca 00 3c 27 b3 98 ba 53 30 37 32 32 34 33 30 00 00 00 00 00 ff 04 01 01 00 00 00 22 04 1e 00 28 00 00 00 00 00 0f 00 f0 03 1c 02 00 00 00 00 46 02 dc 00 78 2d 16 00 01 00 00 00 00 00 00 00 22 4f
7478253 R: 22 23 ba 6c 60 3d 08 01 04 e0 14 07 f4 15
7478254 T: 00 d7 fa 00 1c 10 17 6e 3b 00 30 30 30 30 30 30 30 30 30 30 00 00 00 a9 75 6b 06 00 00 00 00 00 00 10 08
7478271 R: 22 23 ba 6c 60 3d 08 01 04 e0 1e 0c 57 56
7478271 T: 00 32 8c 00 30 ba 47 99 b3 00 00 00 00 ac 87 ff 56 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 67 a2
7478273 R: 22 23 ba 6c 60 3d 08 01 04 e0 32 0b 7b a8
7478274 T: 00 11 f3 00 2c dd c3 3e 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 38 87
7478276 R: 22 b2 04 ba 6c 60 3d 08 01 04 e0 1e f8
7478276 T: 00 5e e8 6d e4
7478276 R: 22 b3 04 ba 6c 60 3d 08 01 04 e0 01 d1 25 0e ee 37 81
7478277 T: 00 78 f0
7478277 R: 22 23 ba 6c 60 3d 08 01 04 e0 4f 01 9d 47
7478278 T: 01 0f 68 ee
7478278 R: 36 01 00 00 6a a1
7478278 T: 00 01 ba 6c 60 3d 08 01 04 e0 5d 2b
7478278 R: 22 23 ba 6c 60 3d 08 01 04 e0 00 03 21 a1
7478279 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
7478280 R: 22 bd 04 ba 6c 60 3d 08 01 04 e0 c2 55
7478280 T: 00 33 4a 63 63 d0 13 49 db a0 9e ee 15 1e f8 f8 f3 fa 15 f5 77 e4 4d 75 9b 78 14 ca d3 7e 02 ef 10 6d 93
7478293 R: 22 2b ba 6c 60 3d 08 01 04 e0 af cc
7478293 T: 00 0f ba 6c 60 3d 08 01 04 e0 01 3d 4f 03 01 40 d3
7478294 R: 22 23 ba 6c 60 3d 08 01 04 e0 00 03 21 a1
7478294 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
7478295 R: 22 23 ba 6c 60 3d 08 01 04 e0 03 0f 25 41
7478295 T: 00 b6 ca 00 3c 27 b3 98 ba 53 30 37 32 32 34 33 30 00 00 00 00 00 ff 04 01 01 00 00 00 22 04 1e 00 28 00 00 00 00 00 0f 00 f0 03 1c 02 00 00 00 00 46 02 dc 00 78 2d 16 00 01 00 00 00 00 00 00 00 22 4f
7478298 R: 22 23 ba 6c 60 3d 08 01 04 e0 14 07 f4 15
7478298 T: 00 d7 fa 00 1c 10 17 6e 3b 00 30 30 30 30 30 30 30 30 30 30 00 00 00 a9 75 6b 06 00 00 00 00 00 00 10 08
7478300 R: 22 23 ba 6c 60 3d 08 01 04 e0 1e 0c 57 56
7491148 T: 30 ba 47 99 b3 00 00 00 00 ac 87 ff 56 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 67 a2 01 00 0e 00 22 23 ba 6c 60 3d 08 01 04 e0 32 0b 7b a8 00 00 33 00 00 11 f3 00 2c dd c3 3e 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 38 87 01 00 0d 00 22 b2 04 ba 6c 60 3d 08 01 04 e0 1e
7491172 T: 3c 27 b3 98 ba 53 30 37 32 32 34 33 30 00 00 00 00 00 ff 04 01 01 00 00 00 22 04 1e 00 28 00 00 00 00 00 0f 00 f0 03 1c 02 00 00 00 00 46 02 dc 00 78 2d 16 00 01 00 00 00 00 00 00 00 22 4f 01 00 0c 00 22 26 ba 6c 60 3d 08 01 04 e0 7d c1 00 00 03 00 00 78 f0 01 00 0c 00 22 26 ba 6c 60 3d 08 01 04 e0 7d c1 00 00 03 00 00 78 f0 01 00 0c 00 22 26 ba 6c 60 3d 08 01 04 e0 7d c1 00 00 03 00 00 78 f0 01 00 0c 00 22 26 ba 6c 60 3d 08 01 04 e0 7d c1 00 00 03 00 00 78 f0 01 00 0c 00 22 26 ba 6c 60 3d 08 01 04 e0 7d c1 00 00 03 00 00 78 f0 01 00 0c 00 22 26 ba 6c 60 3d 08 01 04 e0 7d c1 00 00 03 00 00 78 f0 01
```

eychei, 2023-06-19:
I dumped one of my own tags and added the signature manually. This is the content:

```
Filetype: Flipper NFC device
Version: 3
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic or ISO15693
Device type: ISO15693
# UID is common for all formats
UID: E0 04 01 08 2F 81 D8 FC
# Data Storage Format Identifier
DSFID: 01
# Application Family Identifier
AFI: 3D
IC Reference: 01
# Number of memory blocks, usually 0 to 256
Block Count: 80
# Size of a single memory block, usually 4
Block Size: 04
Data Content: 03 0A 82 ED 86 39 61 D2 03 14 1E 32 B6 CA 00 3C 90 AE DB FD 53 30 39 30 34 39 38 30 00 00 00 00 00 FF 04 01 01 00 00 00 73 06 1E 00 26 00 00 00 00 00 10 00 3A 06 14 04 00 00 00 00 33 04 19 00 10 08 03 00 01 00 00 00 00 00 00 00 00 00 00 00 D7 FA 00 1C 76 C6 91 7B 00 30 30 30 30 30 30 30 30 30 30 00 00 00 F9 0C 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 8C 00 30 4C 15 1F 66 00 00 00 00 D2 46 40 6A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 F3 00 2C DD C3 3E 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FD FF 00 01
# First byte: DSFID (0x01) / AFI (0x02) / EAS (0x04) / PPL (0x08) lock info, others: block lock info
Security Status: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
# Subtype of this card (0 = ISO15693, 1 = SLIX, 2 = SLIX-S, 3 = SLIX-L, 4 = SLIX2)
Subtype: 04
# SLIX2 specific data
# Passwords are optional. If password is omitted, any password is accepted
# This is the card's secp128r1 elliptic curve signature. It can not be calculated without knowing NXP's private key.
Signature: 22 B8 3A 23 9B E5 78 F0 2C DE D3 CD 31 BA 6A 20 F1 9A E7 21 F2 FC CC EE 37 FA 68 E7 7A 29 86 20
Privacy Mode: false
# Protection pointer configuration
Protection pointer: 00
Protection condition: 00
```

This is what I get in the debug.txt:

```
6855109 R: 36 01 00 00 6a a1
6855110 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
6855110 R: 36 01 00 00 6a a1
6855110 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
6855111 R: 36 01 00 00 6a a1
6855111 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
6855111 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
6855112 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
6855113 R: 22 bd 04 fc d8 81 2f 08 01 04 e0 53 36
6855113 T: 00 22 b8 3a 23 9b e5 78 f0 2c de d3 cd 31 ba 6a 20 f1 9a e7 21 f2 fc cc ee 37 fa 68 e7 7a 29 86 20 67 cc
6855115 R: 22 2b fc d8 81 2f 08 01 04 e0 3e af
6855115 T: 00 0f fc d8 81 2f 08 01 04 e0 01 3d 4f 03 01 38 5a
6855116 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
6855116 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
6855117 R: 22 23 fc d8 81 2f 08 01 04 e0 03 0f 3d 10
6855117 T: 00 b6 ca 00 3c 90 ae db fd 53 30 39 30 34 39 38 30 00 00 00 00 00 ff 04 01 01 00 00 00 73 06 1e 00 26 00 00 00 00 00 10 00 3a 06 14 04 00 00 00 00 33 04 19 00 10 08 03 00 01 00 00 00 00 00 00 00 0a 16
6855120 R: 22 23 fc d8 81 2f 08 01 04 e0 14 07 ec 44
6855133 T: 00 d7 fa 00 1c 76 c6 91 7b 00 30 30 30 30 30 30 30 30 30 30 00 00 00 f9 0c 10 00 00 00 00 00 00 00 03 28
6855135 R: 22 23 fc d8 81 2f 08 01 04 e0 1e 0c 4f 07
6855135 T: 00 32 8c 00 30 4c 15 1f 66 00 00 00 00 d2 46 40 6a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f3 b6
6855137 R: 22 23 fc d8 81 2f 08 01 04 e0 32 0b 63 f9
6855138 T: 00 11 f3 00 2c dd c3 3e 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 38 87
6855139 R: 22 b2 04 fc d8 81 2f 08 01 04 e0 8f 9b
6855140 T: 00 b5 e0 14 65
6855140 R: 22 b3 04 fc d8 81 2f 08 01 04 e0 01 a0 ed 18 e9 17 37
6855141 T: 00 78 f0
6855141 R: 22 23 fc d8 81 2f 08 01 04 e0 4f 01 85 16
6855141 T: 01 0f 68 ee
6855142 R: 36 01 00 00 6a a1
6855142 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
6855142 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
6855143 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
6855155 R: 22 bd 04 fc d8 81 2f 08 01 04 e0 53 36
6855156 T: 00 22 b8 3a 23 9b e5 78 f0 2c de d3 cd 31 ba 6a 20 f1 9a e7 21 f2 fc cc ee 37 fa 68 e7 7a 29 86 20 67 cc
6855157 R: 22 2b fc d8 81 2f 08 01 04 e0 3e af
6855157 T: 00 0f fc d8 81 2f 08 01 04 e0 01 3d 4f 03 01 38 5a
6855158 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
6855158 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
6855159 R: 22 23 fc d8 81 2f 08 01 04 e0 03 0f 3d 10
6855160 T: 00 b6 ca 00 3c 90 ae db fd 53 30 39 30 34 39 38 30 00 00 00 00 00 ff 04 01 01 00 00 00 73 06 1e 00 26 00 00 00 00 00 10 00 3a 06 14 04 00 00 00 00 33 04 19 00 10 08 03 00 01 00 00 00 00 00 00 00 0a 16
6855162 R: 22 23 fc d8 81 2f 08 01 04 e0 14 07 ec 44
6855163 T: 00 d7 fa 00 1c 76 c6 91 7b 00 30 30 30 30 30 30 30 30 30 30 00 00 00 f9 0c 10 00 00 00 00 00 00 00 03 28
6855164 R: 22 23 fc d8 81 2f 08 01 04 e0 1e 0c 4f 07
6863064 T: 30 4c 15 1f 66 00 00 00 00 d2 46 40 6a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f3 b6 01 00 0e 00 22 23 fc d8 81 2f 08 01 04 e0 32 0b 63 f9 00 00 33 00 00 11 f3 00 2c dd c3 3e 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 38 87 01 00 0d 00 22 b2 04 fc d8 81 2f 08 01 04 e0 8f

```

eychei, 2023-06-19:
Changed to 0F:
Still not recognized by the Dymo reader:

```
7435759 R: 36 01 00 00 6a a1
7435760 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
7435760 R: 36 01 00 00 6a a1
7435760 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
7435761 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
7435761 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
7435762 R: 22 bd 04 fc d8 81 2f 08 01 04 e0 53 36
7435763 T: 00 22 b8 3a 23 9b e5 78 f0 2c de d3 cd 31 ba 6a 20 f1 9a e7 21 f2 fc cc ee 37 fa 68 e7 7a 29 86 20 67 cc
7435764 R: 22 2b fc d8 81 2f 08 01 04 e0 3e af
7435764 T: 00 0f fc d8 81 2f 08 01 04 e0 01 3d 4f 03 01 38 5a
7435765 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
7435766 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
7435766 R: 22 23 fc d8 81 2f 08 01 04 e0 03 0f 3d 10
7435767 T: 00 b6 ca 00 3c 90 ae db fd 53 30 39 30 34 39 38 30 00 00 00 00 00 ff 04 01 01 00 00 00 73 06 1e 00 26 00 00 00 00 00 10 00 3a 06 14 04 00 00 00 00 33 04 19 00 10 08 03 00 01 00 00 00 00 00 00 00 0a 16
7435769 R: 22 23 fc d8 81 2f 08 01 04 e0 14 07 ec 44
7435770 T: 00 d7 fa 00 1c 76 c6 91 7b 00 30 30 30 30 30 30 30 30 30 30 00 00 00 f9 0c 10 00 00 00 00 00 00 00 03 28
7435822 R: 22 23 fc d8 81 2f 08 01 04 e0 1e 0c 4f 07
7435823 T: 00 32 8c 00 30 4c 15 1f 66 00 00 00 00 d2 46 40 6a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f3 b6
7435825 R: 22 23 fc d8 81 2f 08 01 04 e0 32 0b 63 f9
7435825 T: 00 11 f3 00 2c dd c3 3e 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 38 87
7435827 R: 22 b2 04 fc d8 81 2f 08 01 04 e0 8f 9b
7435828 T: 00 88 68 8e 6b
7435828 R: 22 b3 04 fc d8 81 2f 08 01 04 e0 01 9d 65 25 61 de 34
7435829 T: 00 78 f0
7435829 R: 22 23 fc d8 81 2f 08 01 04 e0 4f 01 85 16
7435829 T: 01 0f 68 ee
7435830 R: 36 01 00 00 6a a1
7435830 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
7435830 R: 36 01 00 00 6a a1
7435830 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
7435831 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
7435831 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
7435844 R: 22 bd 04 fc d8 81 2f 08 01 04 e0 53 36
7435844 T: 00 22 b8 3a 23 9b e5 78 f0 2c de d3 cd 31 ba 6a 20 f1 9a e7 21 f2 fc cc ee 37 fa 68 e7 7a 29 86 20 67 cc
7435845 R: 22 2b fc d8 81 2f 08 01 04 e0 3e af
7435846 T: 00 0f fc d8 81 2f 08 01 04 e0 01 3d 4f 03 01 38 5a
7435846 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
7435847 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
7435848 R: 22 23 fc d8 81 2f 08 01 04 e0 03 0f 3d 10
7435848 T: 00 b6 ca 00 3c 90 ae db fd 53 30 39 30 34 39 38 30 00 00 00 00 00 ff 04 01 01 00 00 00 73 06 1e 00 26 00 00 00 00 00 10 00 3a 06 14 04 00 00 00 00 33 04 19 00 10 08 03 00 01 00 00 00 00 00 00 00 0a 16
7435851 R: 22 23 fc d8 81 2f 08 01 04 e0 14 07 ec 44
7435851 T: 00 d7 fa 00 1c 76 c6 91 7b 00 30 30 30 30 30 30 30 30 30 30 00 00 00 f9 0c 10 00 00 00 00 00 00 00 03 28
7435852 R: 22 23 fc d8 81 2f 08 01 04 e0 1e 0c 4f 07
7459613 T: 30 4c 15 1f 66 00 00 00 00 d2 46 40 6a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f3 b6 01 00 0e 00 22 23 fc d8 81 2f 08 01 04 e0 32 0b 63 f9 00 00 33 00 00 11 f3 00 2c dd c3 3e 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 38 87 01 00 0d 00 22 b2 04 fc d8 81 2f 08 01 04 e0 8f
7459636 T: 3c 90 ae db fd 53 30 39 30 34 39 38 30 00 00 00 00 00 ff 04 01 01 00 00 00 73 06 1e 00 26 00 00 00 00 00 10 00 3a 06 14 04 00 00 00 00 33 04 19 00 10 08 03 00 01 00 00 00 00 00 00 00 0a 16 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01

```

eychei, 2023-06-19:
I checked with the proxmark3 again:
The bits that are different:
Flipper: 
raw : 00 00 00 0F 7F 35 00 00

Original Tag:
raw : 00 32 02 0F 7F 35 00 00

eychei, 2023-06-19:
Edit:
Changed the Protection pointers to 32 02.

The proxmark output is now identical. 
Still not working though.

This is what I get from the debug.txt
```
77562 R: 36 01 00 00 6a a1
77562 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
77563 R: 36 01 00 00 6a a1
77563 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
77563 R: 36 01 00 00 6a a1
77564 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
77564 R: 36 01 00 00 6a a1
77564 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
77565 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
77565 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
77566 R: 22 bd 04 fc d8 81 2f 08 01 04 e0 53 36
77567 T: 00 22 b8 3a 23 9b e5 78 f0 2c de d3 cd 31 ba 6a 20 f1 9a e7 21 f2 fc cc ee 37 fa 68 e7 7a 29 86 20 67 cc
77568 R: 22 2b fc d8 81 2f 08 01 04 e0 3e af
77568 T: 00 0f fc d8 81 2f 08 01 04 e0 01 3d 4f 03 01 38 5a
77569 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
77570 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
77570 R: 22 23 fc d8 81 2f 08 01 04 e0 03 0f 3d 10
77571 T: 00 b6 ca 00 3c 90 ae db fd 53 30 39 30 34 39 38 30 00 00 00 00 00 ff 04 01 01 00 00 00 73 06 1e 00 26 00 00 00 00 00 10 00 3a 06 14 04 00 00 00 00 33 04 19 00 10 08 03 00 01 00 00 00 00 00 00 00 0a 16
77626 R: 22 23 fc d8 81 2f 08 01 04 e0 14 07 ec 44
77627 T: 00 d7 fa 00 1c 76 c6 91 7b 00 30 30 30 30 30 30 30 30 30 30 00 00 00 f9 0c 10 00 00 00 00 00 00 00 03 28
77628 R: 22 23 fc d8 81 2f 08 01 04 e0 1e 0c 4f 07
77629 T: 00 32 8c 00 30 4c 15 1f 66 00 00 00 00 d2 46 40 6a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f3 b6
77631 R: 22 23 fc d8 81 2f 08 01 04 e0 32 0b 63 f9
77631 T: 00 11 f3 00 2c dd c3 3e 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 38 87
77633 R: 22 b2 04 fc d8 81 2f 08 01 04 e0 8f 9b
77633 T: 00 94 57 cb 9e
77634 R: 22 b3 04 fc d8 81 2f 08 01 04 e0 01 81 5a 39 5e 67 59
77634 T: 00 78 f0
77634 R: 22 23 fc d8 81 2f 08 01 04 e0 4f 01 85 16
77635 T: 01 0f 68 ee
77635 R: 36 01 00 00 6a a1
77635 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
77636 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
77636 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
77648 R: 22 bd 04 fc d8 81 2f 08 01 04 e0 53 36
77649 T: 00 22 b8 3a 23 9b e5 78 f0 2c de d3 cd 31 ba 6a 20 f1 9a e7 21 f2 fc cc ee 37 fa 68 e7 7a 29 86 20 67 cc
77650 R: 22 2b fc d8 81 2f 08 01 04 e0 3e af
77651 T: 00 0f fc d8 81 2f 08 01 04 e0 01 3d 4f 03 01 38 5a
77651 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
77652 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
77652 R: 22 23 fc d8 81 2f 08 01 04 e0 03 0f 3d 10
77653 T: 00 b6 ca 00 3c 90 ae db fd 53 30 39 30 34 39 38 30 00 00 00 00 00 ff 04 01 01 00 00 00 73 06 1e 00 26 00 00 00 00 00 10 00 3a 06 14 04 00 00 00 00 33 04 19 00 10 08 03 00 01 00 00 00 00 00 00 00 0a 16
77655 R: 22 23 fc d8 81 2f 08 01 04 e0 14 07 ec 44
77656 T: 00 d7 fa 00 1c 76 c6 91 7b 00 30 30 30 30 30 30 30 30 30 30 00 00 00 f9 0c 10 00 00 00 00 00 00 00 03 28
77657 R: 22 23 fc d8 81 2f 08 01 04 e0 1e 0c 4f 07
91932 T: 00 32 8c 00 30 4c 15 1f 66 00 00 00 00 d2 46 40 6a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f3 b6
91952 R: 22 23 fc d8 81 2f 08 01 04 e0 32 0b 63 f9
91952 T: 00 11 f3 00 2c dd c3 3e 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 38 87
91954 R: 22 b2 04 fc d8 81 2f 08 01 04 e0 8f 9b
91954 T: 00 74 9e 9f 2c
91955 R: 22 b3 04 fc d8 81 2f 08 01 04 e0 01 61 93 d9 97 3d c9
91955 T: 00 78 f0
91956 R: 22 23 fc d8 81 2f 08 01 04 e0 4f 01 85 16
91956 T: 01 0f 68 ee
91956 R: 36 01 00 00 6a a1
91956 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
91957 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
91957 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
91958 R: 22 bd 04 fc d8 81 2f 08 01 04 e0 53 36
91959 T: 00 22 b8 3a 23 9b e5 78 f0 2c de d3 cd 31 ba 6a 20 f1 9a e7 21 f2 fc cc ee 37 fa 68 e7 7a 29 86 20 67 cc
91960 R: 22 2b fc d8 81 2f 08 01 04 e0 3e af
91960 T: 00 0f fc d8 81 2f 08 01 04 e0 01 3d 4f 03 01 38 5a
[... 5 lines cut ...]
91984 T: 00 d7 fa 00 1c 76 c6 91 7b 00 30 30 30 30 30 30 30 30 30 30 00 00 00 f9 0c 10 00 00 00 00 00 00 00 03 28
91985 R: 22 23 fc d8 81 2f 08 01 04 e0 1e 0c 4f 07
91985 T: 00 32 8c 00 30 4c 15 1f 66 00 00 00 00 d2 46 40 6a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f3 b6
91987 R: 22 23 fc d8 81 2f 08 01 04 e0 32 0b 63 f9
91988 T: 00 11 f3 00 2c dd c3 3e 91 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 38 87
91990 R: 22 b2 04 fc d8 81 2f 08 01 04 e0 8f 9b
91990 T: 00 42 a2 02 35
91990 R: 22 b3 04 fc d8 81 2f 08 01 04 e0 01 57 af ef ab c5 f4
91991 T: 00 78 f0
92011 R: 22 23 fc d8 81 2f 08 01 04 e0 4f 01 85 16
92011 T: 01 0f 68 ee
92011 R: 36 01 00 00 6a a1
92011 T: 00 01 fc d8 81 2f 08 01 04 e0 cc 48
92012 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
92012 T: 00 03 0a 82 ed 86 39 61 d2 03 14 1e 32 b6 ca 00 3c d4 c3
92013 R: 22 bd 04 fc d8 81 2f 08 01 04 e0 53 36
92014 T: 00 22 b8 3a 23 9b e5 78 f0 2c de d3 cd 31 ba 6a 20 f1 9a e7 21 f2 fc cc ee 37 fa 68 e7 7a 29 86 20 67 cc
92015 R: 22 2b fc d8 81 2f 08 01 04 e0 3e af
92015 T: 00 0f fc d8 81 2f 08 01 04 e0 01 3d 4f 03 01 38 5a
92016 R: 22 23 fc d8 81 2f 08 01 04 e0 00 03 39 f0
[... cut ...]
92018 T: 3c 90 ae db fd 53 30 39 30 34 39 38 30 00 00 00 00 00 ff 04 01 01 00 00 00 73 06 1e 00 26 00 00 00 00 00 10 00 3a 06 14 04 00 00 00 00 33 04 19 00 10 08 03 00 01 00 00 00 00 00 00 00 0a 16 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01 00 0c 00 22 26 fc d8 81 2f 08 01 04 e0 ec a2 00 00 03 00 00 78 f0 01

```

eychei, 2023-06-19:
This is the log from the proxmark3 sniff:

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |      52224 | Rdr |36  01  00  00  6a  a1                                                   |  ok |
 INVENTORY
    9638208 |    9682240 | Rdr |36  01  00  15  5c                                                       | !crc|
 INVENTORY
   28938144 |   28990368 | Rdr |36  01  00  00  6a  a1                                                   |  ok |
 INVENTORY
   38574304 |   38626528 | Rdr |36  01  00  00  6a  a1                                                   |  ok |
 INVENTORY
   48210400 |   48262624 | Rdr |36  01  00  00  6a  a1                                                   |  ok |
 INVENTORY
   57846528 |   57898752 | Rdr |36  01  00  00  6a  a1                                                   |  ok |
 INVENTORY
   67482592 |   67534816 | Rdr |36  01  00  00  6a  a1                                                   |  ok |
 INVENTORY
   77118656 |   77170880 | Rdr |36  01  00  00  6a  a1                                                   |  ok |
 INVENTORY
   77175008 |   77228256 | Tag |00  01  fc  d8  81  2f  08  01  04  e0  cc  48                           |  ok |

   77277856 |   77395616 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  00  03  39  f0                   |  ok |
 READ_MULTI_BLOCK(0-3)
   77471264 |   77580832 | Rdr |22  bd  04  fc  d8  81  2f  08  01  04  e0  53  36                       |  ok |
 READ_SIGNATURE
   77586720 |   77734176 | Tag |00  22  b8  3a  23  9b  e5  78  f0  2c  de  d3  cd  31  ba  6a  20  f1   |     |

            |            |     |9a  e7  21  f2  fc  cc  ee  37  fa  68  e7  7a  29  86  20  67  cc       |  ok |

   79408992 |   79510368 | Rdr |22  2b  fc  d8  81  2f  08  01  04  e0  3e  af                           |  ok |
 GET_SYSTEM_INFO
   79514528 |   79588256 | Tag |00  0f  fc  d8  81  2f  08  01  04  e0  01  3d  4f  03  01  38  5a       |  ok |

   79820032 |   79864064 | Rdr |22  e3  00  03  39                                                       | !crc|
 Proprietary IC MFG dependent
   79869856 |   79951776 | Tag |00  03  0a  82  ed  86  39  61  d2  03  14  1e  32  b6  ca  00  3c  d4   |     |

            |            |     |c3                                                                       |  ok |

   80012896 |   80130656 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  03  0f  3d  10                   |  ok |
 READ_MULTI_BLOCK(3-18)
   80776832 |   80853632 | Rdr |22  23  fc  d8  81  2f  08  01  c1                                       | !crc|
 READ_MULTI_BLOCK(0-96)
   81513344 |   81622912 | Rdr |22  b2  04  fc  d8  81  2f  08  01  04  e0  8f  9b                       |  ok |
 GET_RANDOM_NUMBER
   81852704 |   82003232 | Rdr |22  b3  04  fc  d8  81  2f  08  01  04  e0  01  03  6b  bb  6f  7b  0e   |  ok |
 SET_PASSWORD
   82168544 |   82286304 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  4f  01  85  16                   |  ok |
 READ_MULTI_BLOCK(79-80)
   87643776 |   87925376 | Rdr |00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00   |     |

            |            |     |00  00  00  00  00  00  00  00  00  00  00  00  00  00  d0  ed           | !crc|
 ?
   89466432 |   89567808 | Rdr |22  2b  fc  d8  81  2f  08  01  04  e0  3e  af                           |  ok |
 GET_SYSTEM_INFO
   89571936 |   89645664 | Tag |00  0f  fc  d8  81  2f  08  01  04  e0  01  3d  4f  03  01  38  5a       |  ok |

   89840352 |   89958112 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  00  03  39  f0                   |  ok |
 READ_MULTI_BLOCK(0-3)
   89962528 |   90044448 | Tag |00  03  0a  82  ed  86  39  61  d2  03  14  1e  32  b6  ca  00  3c  d4   |     |

            |            |     |c3                                                                       |  ok |

   91354144 |   91471904 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  32  0b  63  f9                   |  ok |
 READ_MULTI_BLOCK(50-61)
   91811872 |   91921440 | Rdr |22  b2  04  fc  d8  81  2f  08  01  04  e0  8f  9b                       |  ok |
 GET_RANDOM_NUMBER
   91938592 |   91963168 | Tag |00  21  f2  ba  28                                                       |  ok |
[... 70 lines cut ...]

  200786688 |  200888064 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  200892256 |  200908640 | Tag |00  78  f0                                                               |  ok |

  214313664 |  214415040 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  214424032 |  214440416 | Tag |00  78  f0                                                               |  ok |

  227873344 |  227974720 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  227978848 |  227995232 | Tag |00  78  f0                                                               |  ok |

  241400288 |  241501664 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  241505824 |  241522208 | Tag |00  78  f0                                                               |  ok |

  254960032 |  255061408 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  255065568 |  255081952 | Tag |00  78  f0                                                               |  ok |

  268486880 |  268588256 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok |
 RESET_TO_READY
  268592416 |  268608800 | Tag |00  78  f0                                                               |  ok |
```

g3gg0, 2023-06-19:
hint: when there is a !crc flag, then the reading was probably incorrect. 
maybe you can try to get clean reads to compare them side by side.

interesting is, that i do not see any real multi block response...
only block 0-3 is read properly, the others... not sure what i see

eychei, 2023-06-19:
Ok I will sniff a original tag and then try to get also good readings with the flipper.

eychei, 2023-06-19:
I tried getting reads without any crc errors and data skips without success.
So I copy pasted every line I could find without crc errors for 10-20 runs.

Original Tag in Dymo reader:
```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |      52224 | Rdr |36  01  00  00  6a  a1                                                   |  ok | INVENTORY
      56416 |     109664 | Tag |00  01  fc  d8  81  2f  08  01  04  e0  cc  48                           |  ok |
	  	  
	 389664 |     499232 | Rdr |22  bd  04  fc  d8  81  2f  08  01  04  e0  53  36                       |  ok | READ_SIGNATURE
     503424 |     650880 | Tag |00  22  b8  3a  23  9b  e5  78  f0  2c  de  d3  cd  31  ba  6a  20  f1   |     |
            |            |     |9a  e7  21  f2  fc  cc  ee  37  fa  68  e7  7a  29  86  20  67  cc       |  ok |

    2701856 |    2819616 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  00  03  39  f0                   |  ok | READ_MULTI_BLOCK(0-3)
    2823840 |    2905760 | Tag |00  03  0a  82  ed  86  39  61  d2  03  14  1e  32  b6  ca  00  3c  d4   |     |
            |            |     |c3                                                                       |  ok |
	  
    2333184 |    2434560 | Rdr |22  2b  fc  d8  81  2f  08  01  04  e0  3e  af                           |  ok | GET_SYSTEM_INFO
    2438752 |    2512480 | Tag |00  0f  fc  d8  81  2f  08  01  04  e0  01  3d  4f  03  01  38  5a       |  ok |	  
	  

     159136 |     276896 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  00  03  39  f0                   |  ok | READ_MULTI_BLOCK(0-3)
     281056 |     362976 | Tag |00  03  0a  82  ed  86  39  61  d2  03  14  1e  32  b6  ca  00  3c  d4   |     |
            |            |     |c3                                                                       |  ok |
			
  171410400 |  171528160 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  14  07  ec  44                   |  ok | READ_MULTI_BLOCK(20-27)
  171532352 |  171679808 | Tag |00  d7  fa  00  1c  76  c6  91  7b  00  30  30  30  30  30  30  30  30   |     |
            |            |     |30  30  00  00  00  f9  0c  10  00  00  00  00  00  00  00  03  28       |  ok |	  
	  
    4378560 |    4488128 | Rdr |22  b2  04  fc  d8  81  2f  08  01  04  e0  8f  9b                       |  ok | GET_RANDOM_NUMBER
    4492288 |    4516864 | Tag |00  26  39  6d  1d                                                       |  ok |
	
    4825184 |    4975712 | Rdr |22  b3  04  fc  d8  81  2f  08  01  04  e0  01  33  34  8b  30  7d  97   |  ok | SET_PASSWORD
    4979904 |    4996288 | Tag |00  78  f0                                                               |  ok |	  
	  
    5202400 |    5320160 | Rdr |22  23  fc  d8  81  2f  08  01  04  e0  4f  01  85  16                   |  ok | READ_MULTI_BLOCK(79-80)
    5324320 |    5357088 | Tag |00  fd  ff  00  01  a9  e4                                               |  ok |

   35594880 |   35696256 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok | RESET_TO_READY
   35700448 |   35716832 | Tag |00  78  f0                                                               |  ok |
   49154592 |   49255968 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok | RESET_TO_READY
   49260128 |   49276512 | Tag |00  78  f0                                                               |  ok |
   62681504 |   62782880 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok | RESET_TO_READY
   62787104 |   62803488 | Tag |00  78  f0                                                               |  ok |
   76241216 |   76342592 | Rdr |22  26  fc  d8  81  2f  08  01  04  e0  ec  a2                           |  ok | RESET_TO_READY
   76346784 |   76363168 | Tag |00  78  f0                                                               |  ok |
```

g3gg0, 2023-06-19:
the reason: 
reader reads blocks 79 *and* 80 (which is one behind) and the tag responds with "naaah, not possible".
the SLIX2 spec says in this case it will respond with only the valid blocks.
so does some other SLIX-L tag...

[image]

```
hf 15 raw -c -d 4223xxyy
xx: start block
yy: block count minus one
```

will fix this in a minute

eychei, 2023-06-19:
Yes. Will try now.

eychei, 2023-06-19:
Did you push the changes?

eychei, 2023-06-19:
IT WORKS!!!
Great Work!
It does recognize all 3 Tags (two from you and one by me).
I did print a page with the dymo and it also did write the counter value (decrement by one) without issues.
I also tried the restore function and by doing so I can reset the counter to the initial value.

It is not a cheap solution (buying a flipper) to use 3rd party labels for the dymo printer, but it may be useful for someone else in the future.

Thank you.

-e

g3gg0, 2023-06-19:
great to read that it worked.
thanks for testing :)

g3gg0, 2023-06-19:
can you check if the just pushed commit adds SLIX2 readout properly?

i cannot test it, as i neither have a SLIX2 tag nor a second flipper.
the pm3 doesn't allow emulating SLIX2 yet.

eychei, 2023-06-19:
I can see that you pushed new code. Will compile the firmware tomorrow and give feedback. Could be late evening though. 

-e

eychei, 2023-06-20:
Hi,

tested the new version and it works out of the box. But some info is not read and copied. Maybe this is needed for other systems:

1. Security Status is Empty, Maybe the Page Protection is also not read
# First byte: DSFID (0x01) / AFI (0x02) / EAS (0x04) / PPL (0x08) lock info, others: block lock info
Security Status: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

2. Protection pointer are not set
# Protection pointer configuration
Protection pointer: 00
Protection condition: 00

-->
 --------- NXP Sysinfo ---------
Original Tag:
raw : 00 32 02 0F 7F 35 00 00

3. It would also be nice to have the flipper zero store the PASSWORDS when the reader is sending out these.
That way we would not need to sniff these when using original tags. 

-e

g3gg0, 2023-06-20:
was the signature read correctly?

passwords are not known, which ones to store?

g3gg0, 2023-06-20:
or do you mean when emulating, the password gets updated in the .nfc file?

eychei, 2023-06-20:
Yes signature was read correctly. I tried two tags. Just cloned them and emulated. They worked fine on the Dymo printer.

The Dymo printer is sending the write password so to increment the counter later on.
Can you store this in the flipper during emulation?

eychei, 2023-06-20:
> or do you mean when emulating, the password gets updated in the .nfc file?

Yes exactly.

g3gg0, 2023-06-20:
another update, can you try please?

eychei, 2023-06-21:
Hi. 
Tried the update.
The Security Status and the Protection pointers are read correctly now.

Did you also implement the password update function while emulating the tag?

-e

g3gg0, 2023-06-21:
great!
nah, didn't add this feature yet. but added right now.

can you check if that works as expected?

g3gg0, 2023-06-21:
current state:
it should read a SLIX2 and be able to emulate it.
all information neeeded will be read, *except* the security status bits for every block. (is this even required?)

@gornekich @nvx ISO15693-3 10.4.1 states that if Option_flag is set, the VICC responds with the block security status.
do you think it is okay to change nfcv_read_blocks() to always add this flag and handle the status byte, if it was returned?
a) i don't want to make an extra READ BLOCK (option 1) readout function just for reading the status byte
b) using "10.4.13 Get multiple block security status" commands would just work with SLIX2, as SLIX-L only supports READ BLOCK (option 1)
c) not implementing it feels wrong :)

eychei, 2023-06-21:
Tried newest firmware and the password is stored now. Great work!
But if I read one tag get the password and read a second tag afterwards, the password of the first tag is also stored in the second tag. 
The password field and all other fields needs to be cleared on every new read, else the data of the old tag is used. Can you check that ?

eychei, 2023-06-21:
Just found another bug. 
mod you check the tag content on the flipper it will only show you 50 blocks. The tag does have 80 blocks though.

g3gg0, 2023-06-21:
password storing fixed, cah you check again?

[image]
it's intentional to not show too much information in that small textbox, if you meant that?

eychei, 2023-06-21:
I meant the Block size field which is shown on the flipper screen when pressing the info button.
It says 50 blocks but the data is 80 blocks in the nfc file.
Also scrolling down through the info page on the flipper only 50 blocks are shown. I can not scroll down to see the counter value for example.

g3gg0, 2023-06-21:
good point, this is shown in hexadecimal, but not annotated. will fix it.

eychei, 2023-06-21:
Ok tested the newest version. 
Reading Tags is working now as expected. The Password field and others are not copied over from the old tag data.
Great Work!

Will test more if you have new updates.

g3gg0, 2023-06-21:
thanks for your extensive testing and technical support.

yeah, use git <at> g3gg0.de

eychei, 2023-06-21:
How is this working over git? New to this thing for chating or sending messages.

g3gg0, 2023-06-21:
fixed the message. just mail me.

closed 2023-11-01 by ImagineVagon333

Morfis1855, 2024-03-10:
hi there
I had a similar issue but I didn't have a programming Background, I'm working in a medical lab, two yeas ago I had purchased an inventory system the manufacturer provided a software plus hardware (readers) and 5 years licence we started using those inlay tags which are expensive but tolerable cost after one year they start raising the price for each tag till reaching more than 4$ now which is insanely high I believe you had an idea about how much inlay Slix cost from china or Europe suppliers which is less than 1 $, I got stuck with expensive hardware and 5 years prepaid licence Plz help to crack those tags, I had punch of Nxp Slix tags from different supplier and a PM3 all my trials to clone a serial of those tags went south even if NXP Info and/or PM3 show them as identical to the original

g3gg0, 2024-03-10:
> hi there I had a similar issue but I didn't have a programming Background, I'm working in a medical lab, two yeas ago I had purchased an inventory system the manufacturer provided a software plus [...]

No offense, but please restucture your post a bit using interpuction and newlines.
It is too hard to follow what your point really is.

Morfis1855, 2024-03-10:
Sorry for my English 

hi there.

I had a similar issue but I do not  have a programming Background, I'm working in a medical lab, two yeas ago had purchased an inventory system to track samples and reagents across the lab using RFID Labels. 
the manufacturer provided a software, hardware (readers) and 5 years licence.

we started to use those inlay REID tags which are expensive ~2$ but tolerable cost.
after one year the supplier started raising the price for each tag till reaching more than 4$ last month, which is insanely high.
 I believe you had an idea about how much RFID lable cost from china or Europe  it cost less than 1 $, Now I got stuck with expensive hardware and 5 years prepaid licence Plz help to crack those tags,

I bought 50 sticker from aliexpress @15$ with the same exact IC "NXP-ICODESLIX, tried to clone But didn’t work.
All my trials to clone a serial of those tags went south even if NXP, MCT Apps and/or PM3 but  reader failed to see tags at all

I HAD attached screenshot of both original tag and out sourced tags scanning results 

B.R
Morfis 
[image]

[image]

[22 more replies]