flipperdevices/flipperzero-firmware #2789: Add bluetooth shutter button to flipper

geks52, 2023-06-21:
### Description of the feature you're suggesting.

Hello!
Could you please tell me if it is technically possible to write an application to control the camera on mobile devices?
I would like to connect the flipper to my mobile phone, run the camera on it and take photos using the flipper.

Such devices exist, but are sold separately.
[Example.](https://www.amazon.com/bluetooth-remote-iphone-camera/s?k=bluetooth+remote+for+iphone+camera)
[image]

It would be cool to add such a feature!
Thanks for listening!

skotopes, 2023-06-22:
Already in `Bluetooth Remote` application. Media keyboard should do the thing.

closed 2023-06-22 by skotopes

geks52, 2023-06-22:
@skotopes 
Thank you!
The problem was with the Google Camera app.
It is in this app that the flipper photo capture doesn't work. (Pixel 4a, Android 13)
If you install another photo shooting app, everything works.