flipperdevices/flipperzero-firmware #2796: Flipper unable to read access card

DamianCPH, 2023-06-22:
### Describe the bug.

I have an access card for work that I have tried to read, its RFID and I have been only able to read the Raw data, I'm unsure how to view this or what to do with it and I've read I can report it as an issue that maybe one of you can help, it would be really appreciated if you could as I'd like to reproduce this as a key fob rather than card.
[RfidRecord.ask.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11839309/RfidRecord.ask.zip)

### Reproduction

N/A

### Target

RFID Access Card

skotopes, 2023-06-27:
Doesn't looks like RFID. Any IDs on the tag or reader?
Also do you have proxmark?

DamianCPH, 2023-06-27:
Nope no proxmark or markings on the card, wouldn't pick up on NFC only RFID!

skotopes, 2023-06-27:
How about picopass app?

DamianCPH, 2023-06-27:
Admittedly I've never heard of that? Can it be got in the play store?

skotopes, 2023-06-27:
Picopass app is a part of the firmware, can be found in `Applications > NFC`

DamianCPH, 2023-06-27:
Thank you for the explanation, nope doesn't even pick up the card :(

skotopes, 2023-06-27:
We'll release new reader detector app in 0.86, wait for release and try to detect reader.
As a last resort try to find proxmark

DamianCPH, 2023-06-27:
Perfect thank you!

DamianCPH, 2023-06-27:
Hello,

Just after doing some asking around I've discovered the cards that I've
been unable to read are Legic brand cards, is this something that could
help me identify how to clone it with my flipper?

Is this format supported? do ye plan on doing it if not

Kind regards,
Damian

*Kind Regards*

*Damian Coleman (he/him)*

DamianColemanPhotography | Freelance Photography / Videographer.

M +353 85 7497230 | E ***@***.***

W damiancolemanphotography.com

DamianColemanPhotography | Cork | Ireland

Facebook <https://www.facebook.com/DamianColemanPhotography/> | Instagram
<https://www.instagram.com/damiancolemanphotography/>| LinkedIn
<https://www.linkedin.com/in/damiancolemanphotography/> | Twitter
<https://twitter.com/DamianColemanPh> | Youtube
<https://www.youtube.com/channel/UC5ksEk9_mkbKdSbrcpnKMDw>

*P* *Please consider the environment before **printing this* *email or its
attachments.*

skotopes, 2023-06-27:
Not supported yet, but eventually will be added.