flipperdevices/flipperzero-firmware #2805: [NFC] Flipper Emulation stopped working on BAS-IP BME-03 after enabling reader's security profile

mishamyte, 2023-06-26:
### Describe the bug.

There is an object with access system built on top of BAS-IP panels (BME-03 reader) + U-Prox readers (U-Prox SE mini + U-Prox SL mini). Mifare 1K tags were used. Initially the whole system worked only with tag UID. For that moment emulation of tag worked fine at all present readers.

Then the decision were taken to increase the security level. For that purposes new MFP tags were bought and all reader's settings were updated: added MFP profile and (what is important) MFC profile was added (for backward compatibility of all issued tags). 

The one key is used for all issued MFC tags, all sectors are protected by it (both A + B keys).  Reader tries to authenticate with that key for all sectors sequentially until auth will be successful.  Filter only against MF Zero is present.

After applying of those changes Flipper's emulation stopped working on the panels with BME-03 reader (and it is working on all other readers). All other tags (like MF-3, multiple versions of chinese CUID, Gen 4 GTU card) are working fine.

I made the dumps by Proxmark3, visually it looks like protocol executes fine. So I have a suggestion it could be a hardware error. 
But before that I decided to create that bug report for checking is it not a software problem. 

Thanks!

### Reproduction

1. Open saved tag (full decrypted with a know key, all sectors are read successfully)
2. Emulate it
3. Try to authenticate via panel with BAS-IP BME-03 reader

_Expected result:_
Door will be opened

_Actual result:_
Door is not opened

Proxmark3 logs are attached:
[proxmark3-traces.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11875049/proxmark3-traces.zip)

### Target

NFC

### Anything else?

**Firmware version:** 0.85.2
_Dump file could be shared securely if needed_

AloneLiberty, 2023-07-14:
There are some improvements in MFC emulation in latest dev release, could you test again on it?

mishamyte, 2023-07-14:
@AloneLiberty I watched #2825, unfortunately improvements from that PR didn't help

mishamyte, 2023-07-14:
Just rechecked with 0.86.2, unfortunately, the problem still present

AloneLiberty, 2023-07-14:
0.86.2 doesn't have my fixes included, try on dev version

mishamyte, 2023-07-14:
Sorry, my bad. Installed `Dev f2324e4d`, still no result

Astrrra, 2023-08-01:
Hm, I think I have an idea why that is happening, but this will probably have to wait until we finish the NFC refactoring

You can try the 0.87.0 release just in case it may work, but if it doesn't, then you'll have to wait for some time

mishamyte, 2023-08-01:
> You can try the 0.87.0 release just in case it may work, but if it doesn't, then you'll have to wait for some time

Hi, I checked. It didn't work.
No worries, will wait for that refactoring

Thanks in advance!

mishamyte, 2023-10-30:
Still reproducible in `[0.94.1-rc](https://github.com/flipperdevices/flipperzero-firmware/releases/tag/0.94.1-rc)`

skotopes, 2024-01-10:
how about latest dev? there were couple emulation fixes

mishamyte, 2024-01-10:
@skotopes rechecked on multiple readers. Still reproducible on `Dev 7eeb60e1`

skotopes, 2024-07-07:
@mishamyte how about now?

mishamyte, 2024-07-07:
Hey @skotopes,

Unfortunately, I could confirm the issue is still valid on `Dev 916af29`

Wondering could it potentially be the issue with the antenna power?

skotopes, 2024-07-07:
@gornekich @doomwastaken looks like the ball is on your side then.