flipperdevices/flipperzero-firmware #2815: [RFID] Electronic RFID hotel door lock system support

valik-svs, 2023-06-29:
### Describe the enhancement you're suggesting.

### Description of the feature you're suggesting.

I've added the raw file of the card.
My flipper zero doesn't seem to be reading RFID at all.
Please help me with my apartment door.
The door lock looks like this:
[image]
card: 
[image]

raw files:
[rfid_raw.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11905315/rfid_raw.zip)

skotopes, 2023-06-29:
Most likely this is NFC based lock. Check NFC application.
Also no meaningful data on raw files your provided

valik-svs, 2023-06-29:
looks like it's my mistake. 

I uploaded corrects files 
[Archive.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11906522/Archive.zip)
[Archive 2.zip](https://github.com/flipperdevices/flipperzero-firmware/files/11906520/Archive.2.zip)

P.S.: `Most likely this is NFC based lock` - I tried to detect the card reader - no result.
Also, try to read via NFC reader - no effect.

skotopes, 2023-06-29:
There is a new app `NFC/RFID reader detect` available in dev build, can be found in Tools. Try using it against reader to detect which carrier it uses.

doomwastaken, 2023-06-29:
google image shows its a  OZLOCKS HL-F05/H/TM Mifare

valik-svs, 2023-06-30:
> There is a new app `NFC/RFID Detector` available in dev build, can be found in Tools. Try using it against reader to detect which carrier it uses.

I checked, it's LF RFID

skotopes, 2023-10-12:
Reader talk first most likely. we need that system in the office for analysis.

closed 2024-07-07 by skotopes