flipperdevices/flipperzero-firmware #2830: USB keyboard utility can't hold down shift

onnimonni, 2023-07-02:
### Description of the feature you're suggesting.

Hey!

Today I didn't have access to usb keyboard when resetting my old Windows PC before selling it.

To reset Windows machine one needs to be able to hold down SHIFT key and press restart button with mouse to access the factory reset screen.

Could we allow the keyboard to emulate that if the shift modifier is selected?

Currently if I press modifier down on flipper it doesn't send the modifier down command to the Operating System.

### Anything else?

Would be fantastic to have new software which would allow me to connect with Qflipper to the flipper and use my Laptop keyboard to send key presses through bluetooth to flipper which would forward them to the PC.

skotopes, 2023-08-01:
Do I understand correctly, we are talking about USB/Bluetooth Remote app?

onnimonni, 2023-08-01:
Yes sorry for being unclear here. I was talking about the USB app here.

onnimonni, 2023-08-01:
>Would be fantastic to have new software which would allow me to connect with Qflipper to the flipper and use my Laptop keyboard to send key presses through bluetooth to flipper which would forward them to the PC.

And this was a separate convenience idea about that if I could connect the flipper to a separate PC as usb-keyboard and then control that keyboard through bluetooth from laptop running the Qflipper software

skotopes, 2023-08-01:
> Yes sorry for being unclear here. I was talking about the USB app here.

We'll fix it.

> And this was a separate convenience idea about that if I could connect the flipper to a separate PC as usb-keyboard and then control that keyboard through bluetooth from laptop running the Qflipper [...]

That will require major redesign in different components. We don't plan to do that at this point.

onnimonni, 2023-08-01:
> We'll fix it.
Well I'm unsure how and would need feedback from you in order to do so.

It's quite clear in the title what the issue was about (USB keyboard) and it's hard to say how different readers will see this.

Now that you have the proper understanding about the issue how would you write it? I don't know what to improve there yet.

> That will require major redesign in different components. We don't plan to do that at this point.
I understand and this was probably not useful to add here in the first place.

How I read the PR template is that you would want to extra feedback/ideas like the ones I provided.

If you don't want to receive stuff like this I would remove following section in the PR template:
```
### Anything else?
```

skotopes, 2023-08-02:
`We'll fix it.` -> `We will fix it`. https://www.merriam-webster.com/dictionary/we'll
@nminaylov is already working on it.

`Anything else` section was meant to be used for additional related to the issue comments.

skotopes, 2024-01-10:
@nminaylov any updates

PR #3378 (2024-02-09): HID app: keyboard modifiers fix
# What's new

- Fixed modifier keys behavior  and added toggled modifiers indication (fixes #2830)
- Small redesign

# Verification 

- Check keyboard function in BLE Remote and USB Remote apps

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix
closed 2024-02-09 by skotopes