flipperdevices/flipperzero-firmware #2834: Picopass app key change to change iCE to elite KDF instead of standard KDF

Akury83, 2023-07-03:
### Description of the feature you're suggesting.

Currently, the Picopass app has a key change feature where the app changes the key on the card.

On the iCE selection, the key is changed to the same key as the iCL key, both using the standard KDF, basically the same raw key is written in either case.

The DRM for the iCE key is the same key as iCL, however it is meant to be encoded using the elite KDF.

Could I please request that the iCE selection uses the elite KDF to change the key when selected?

### Anything else?

As an example, for CSN 2069E013FEFF12E0, the data written in either case when selecting iCE or iCL is 10D7B7245E363973, but should actually be 06E1C3879DBB030E when iCE is selected.

Thank you for all of your hard work on coding this everyone, @bettse in particular. I really appreciate it!!

bettse, 2023-07-03:
ohhhh.  I kept wondering why the keys were the same.  I was following https://github.com/iCopy-X-Community/icopyx-teardown/blob/master/tags/README.md#ice and I see how the `--elite` for ICE that isn't present with the others.  
This should be fixable, but isn't trivial.  I'll look into it when I have some time.

bettse, 2023-07-04:
@Akury83 Give https://github.com/flipperdevices/flipperzero-firmware/pull/2836 a try and see if that fixes it

Akury83, 2023-07-04:
All sorted, thank you @bettse, you're a legend!

closed 2023-07-04 by Akury83