flipperdevices/flipperzero-firmware #2838: Frequencies allowed for Costa Rica

ngonzalez456, 2023-07-04:
### Describe the enhancement you're suggesting.

As can be read here: [http://www.pgrweb.go.cr/scij/Busqueda/Normativa/Normas/nrm_texto_completo.aspx?nValor1=1&nValor2=53219](url), Art. 7, there are no reserved frequencies in the space of 433AM which most of the controllers work around the country. This frequency is not even reserved for AM broadcast. 

Another thing here: [http://www.pgrweb.go.cr/scij/Busqueda/Normativa/Normas/nrm_texto_completo.aspx?param1=NRTC&nValor1=1&nValor2=65675&nValor3=94853&strTipM=TC](url), Addendum VII, and here [http://www.pgrweb.go.cr/scij/Busqueda/Normativa/Normas/nrm_texto_completo.aspx?nValor1=1&nValor2=53219](url), Art. 37, there is no need to validate the device or ask for bandwidth when using less than 200mW (no external antena) or 2W (external antena), less than 200 meters radius and less than 24dBm transmitting power since they will be 'sporadic' transmissions.

I know legal stuff and regulations is a whole thing and I might not be correct, but this is my shot to use any remote in Costa Rica and start learning as I did some time ago in my university.

Thanks.

skotopes, 2023-08-01:
I see `430 - 440` band is being allocated to `RADIO LOCATION`, that make impossible to cover it with amateur use case.
As for `ARTICLE 37` is covering only 2.4G ISM band.

There may be some other documents covering low power, low duty cycle subghz transmission.
Please find them and I'll update CR data.

skotopes, 2024-01-10:
@ngonzalez456 any updates? Should I forward whole thing to certification lab?

ngonzalez456, 2024-01-10:
It's basically imposible to get answers or even the documentation on
anything else than Spanish here, so that's honestly all I could get.