flipperdevices/flipperzero-firmware #2859: NFC: Copy tag content to empty tags instead of initial card

akaash00, 2023-07-10:
### Description of the feature you're suggesting.

Is it possible to write a saved nfc tag to a new empty tag. This way I can make several tags with the same data content. Now you can only write to the initial card. Some systems don’t look at the uid, but look at the data content.

zhiyan114, 2023-07-15:
Fairly sure this is gonna get worked on after refactoring the source

skotopes, 2023-07-31:
Yep, this one planned after NFC refactoring

FatherDivine, 2024-03-06:
Any updates on this? I searched refactoring and only saw NFC posts that mentions issues "after refactoring" occurred. On my flipper, I can only copy to initial and not empty tags. Also can't write to rewritable ntag215's.

skotopes, 2024-07-07:
@FatherDivine refactoring happened around 0.80-0.90. 

@Astrrra @gornekich anything stopping us from implementing it?

ihorman, 2024-07-23:
Please, kindly ask you to add this capability, it is super important