flipperdevices/flipperzero-firmware #2863: ISO15693 emulation not working

nothsa, 2023-07-11:
### Describe the bug.

Trying to emulate a ISO15693 fob, but it is unable to emulate successfully.

When using the `NFC -> Extra Actions -> Listen NfcV Reader` function, I see a number of "`INVENTORY`" lines, and a "`unsupported 07`" line

### Saved NFC file:
```Filetype: Flipper NFC device
Version: 3
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic or ISO15693
Device type: ISO15693
# UID is common for all formats
UID:  [[ REDACTED ]]
# Data Storage Format Identifier
DSFID: 00
# Application Family Identifier
AFI: 30
IC Reference: 8B
# Number of memory blocks, usually 0 to 256
Block Count: 64
# Size of a single memory block, usually 4
Block Size: 04
Data Content: 1A 1A 00 00 51 F5 05 5D 39 AC 0A F5 14 10 44 F6 21 C4 2F 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
# First byte: DSFID (0x01) / AFI (0x02) / EAS (0x04) / PPL (0x08) lock info, others: block lock info
Security Status: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
# Subtype of this card (0 = ISO15693, 1 = SLIX, 2 = SLIX-S, 3 = SLIX-L, 4 = SLIX2)
Subtype: 00
# End of ISO15693 parameters
```
### Debug.txt file:
```60107 R: 07 01 00 11 53
60107 R: 07 01 00 11 20 1c 04 00 44 4c
60107 R: 07 01 00 11 53
60108 R: 07 01 00 11 53
60108 R: 07 01 00 11 53
```

### Reproduction

1. Read ISO15693 fob
2. Select "Emulate"
3. Present the flipper to the scanner

### Logs

```Text
### Saved NFC file:
: Flipper NFC device
Version: 3
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic or ISO15693
Device type: ISO15693
# UID is common for all formats
UID:  [[ REDACTED ]]
# Data Storage Format Identifier
DSFID: 00
# Application Family Identifier
AFI: 30
IC Reference: 8B
# Number of memory blocks, usually 0 to 256
Block Count: 64
# Size of a single memory block, usually 4
Block Size: 04
Data Content: 1A 1A 00 00 51 F5 05 5D 39 AC 0A F5 14 10 44 F6 21 C4 2F 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
# First byte: DSFID (0x01) / AFI (0x02) / EAS (0x04) / PPL (0x08) lock info, others: block lock info
Security Status: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
# Subtype of this card (0 = ISO15693, 1 = SLIX, 2 = SLIX-S, 3 = SLIX-L, 4 = SLIX2)
Subtype: 00
# End of ISO15693 parameters

### Debug.txt file:
```60107 R: 07 01 00 11 53
60107 R: 07 01 00 11 20 1c 04 00 44 4c
60107 R: 07 01 00 11 53
60108 R: 07 01 00 11 53
60108 R: 07 01 00 11 53
```
```

Lukramancer, 2023-07-13:
Same here, even after update to 0.86.2

skotopes, 2023-07-31:
NFC-V support is currently incomplete, we'll come back to it after NFC refactoring. Meanwhile there are people in community who are working on it.

pyllyukko, 2023-08-31:
For me when reading an NFC-V tag, it has the wrong byte order. So it's reversed as in real UID: `00 11 22 33 44 55 66 77` and the Flipper stored it as `77 66 55 44 33 22 11 00`.

doomwastaken, 2023-11-03:
@nothsa @pyllyukko could you check again on latest firmware?

pyllyukko, 2023-11-03:
> @nothsa @pyllyukko could you check again on latest firmware?

I tried it earlier this week (probably with 0.94.0-rc) and it worked perfectly.

doomwastaken, 2023-11-03:
Then it should be fine on 0.94.1 release, thank you. I will wait for other response and will close this issue later

closed 2023-11-08 by doomwastaken