flipperdevices/flipperzero-firmware #2870: Enhanced Frequency Support for India Region

AmeyaVS, 2023-07-12:
### Describe the enhancement you're suggesting.

Currently the default restriction for India Region lies:

[image]

But as per the documentation available [here](https://dot.gov.in/spectrummanagement/rules-use-low-power-radio-frequency-devices-frequency-band-43305-43479-mhz), [pdf](https://dot.gov.in/sites/default/files/Rules%20for%20Use%20of%20Low%20Power%20Radio%20Frequency%20Devices%20in%20the%20frequency%20band%20433-434%20MHz%20(Exemption%20from%20License)%20Rules%2C%202022.pdf?download=1).

Snapshot attached from the document mentioned:
[image]

Kindly enable the 433.05-434.79 MHz band for this region.

skotopes, 2023-08-02:
I've updated India region data. Please reinstall firmware to apply new configuration.

closed 2023-08-02 by skotopes

AmeyaVS, 2023-08-02:
It seems the other frequency are disabled now for the India Region.
This request was to enable additional frequency support for 433 MHz band.

skotopes, 2023-08-03:
@AmeyaVS India doesn't allow those bands for unlicensed use.

AmeyaVS, 2023-08-03:
The general understanding is using other bands is that one does not exceed certain transmission and interference clause.
Their are devices available in the market in the 300+/900 MHz range for very small range applications.

skotopes, 2023-08-04:
@AmeyaVS it explicitly says `in the frequency band as specified in the Table given below`

AmeyaVS, 2023-08-04:
I am not sure why the Indian Government Body have separate documents for every band.
The document linked above is specifically calling out the 433 MHz band.
https://dot.gov.in/spectrummanagement/rules-use-low-power-radio-frequency-devices-frequency-band-43305-43479-mhz